lukevp/ESC-POS-.NET #2: Paper status whith ESCPOS

juliogamasso, 2019-03-11:
good night lukevp, please do you have forecast or could you help me implement return status printer using ESCPOS, I am trying to implement but without success. Thank you!

lukevp, 2019-03-12:
Hi julio,

Which call are you trying to use? the on-demand status call or the other ones?
I haven't implemented any calls that read data back from the printer, but this was on my roadmap for the next release.  Let me know which statuses you're trying to get and I'll make sure they work.

What hardware are you using? 

How is the library working for you other than this missing feature?

juliogamasso, 2019-03-12:
@lukevp, thank you for your attention. The library is fantastic, it caters very well. At the moment, I need to print purchase receipts, but it is crucial to get printer paper status. Our goal is the Epson printer now. I tried to use the command {0x10, 0x04, 0x04} (ASCII = GS r n), but the return is empty.

lukevp, 2019-03-13:
@juliogamasso  I am so glad you like the library!

I have added support for printer statuses.  I have only tested this with the serial printer, so if yours is connected a different way, please let me know if you have any issues.  Here is how you would use this new feature.  You can either subscribe to the status changed event, or you can just check the PrinterStatus public property on the BasePrinter object whenever you need to know the status (it will be null before the first status update comes from the printer).
I have published the updated library to nuget version 0.2.  Please note that the emitter has been renamed EPSON.

`
        static void StatusChanged(object sender, EventArgs ps)
        {
            var status = (PrinterStatus)ps;
            Console.WriteLine($"Printer Online Status: {status.IsPrinterOnline}");
            Console.WriteLine(JsonConvert.SerializeObject(status));
        }

        static void Main(string[] args)
        {
            ICommandEmitter e = new EPSON();
            BasePrinter printer = new SerialPrinter("COM20", 115200);
            printer.StartMonitoring();
            printer.Write(e.Initialize());
            printer.Write(e.Enable());
            printer.Write(e.EnableAutomaticStatusBack());
            printer.StatusChanged += StatusChanged;
        }`

juliogamasso, 2019-03-14:
Hello, @lukevp. I took the test and it worked perfectly. We really like your approach in implementing the ESC / POS standard. We would like to thank you for your support and help.

We are from Ribeirão Preto - São Paulo, Brazil
Thank you!

lukevp, 2019-03-15:
Thank you for the kind words!  I will close this enhancement as it is now resolved.  Please let me know what else you plan to use from the library.

I am from Austin, TX.

closed 2019-03-15 by lukevp