lukevp/ESC-POS-.NET #3: add support for image printing

lukevp, 2019-03-15:


juliogamasso, 2019-04-24:
Hello @lukevp , some prediction of when image printing will be implemented, I have an implementation but its simpler and with .net standart, I tried to adapt mine using its package but I did not succeed. I can not use BitmapImage type in .net standart. Thank you!

lukevp, 2019-04-26:
from @juliogamasso
Hello @lukevp , some prediction of when image printing will be implemented, I have an implementation but its simpler and with .net standart, I tried to adapt mine using its package but I did not succeed. I can not use BitmapImage type in .net standart. Thank you!

lukevp, 2019-05-02:
@juilogamasso
I have added printing in v. 0.4.  Can you please give ESCPOS_NET.ConsoleTest a try and print both the Receipt as well as the Images test (Options 5 and 6)?

If you confirm it is working for you I will close this issue as complete.

lukevp, 2019-05-02:
Also, note that it uses netstandard 2.0 libraries and supports the image formats JPG, PNG, GIF and BMP.

lukevp, 2019-05-02:
@juliogamasso 
It has also been published to Nuget as version 0.4.0 here: https://www.nuget.org/packages/ESCPOS_NET/
It typically takes a few hours to scan and update.

montyclt, 2019-05-02:
@lukevp I will test it this night. Tomorrow I'll write here the test result.

juliogamasso, 2019-05-02:
Hello @lukevp, thank you for tagging me in the message. I tested the Epson TM88v printer and it works perfectly, I have another printer model that uses the ESC-POS protocol but I did not succeed in image printing when sending an image is printing strange characters. I think the problem is related to the printer codepage but I can not find the default codepage for epson. Thank you so much again!

lukevp, 2019-05-02:
I'm glad the TM88V worked! Could you send a picture of the printout from the printer that didn't work? What model is it?
I don't believe that the code page command is implemented yet, but if you let me know what it is I can add that command as well.

juliogamasso, 2019-05-02:
@lukevp 
I have an implementation that I adptei of a git project that works on all the printers I have, this model that does not work is from a national manufacturer "Bematech 4200 TH" it supports the ESCPOS commands but in the test it prints the characters. My implementation has been somewhat extensive, yours is much more summarized and direct. If you do not mind I could send you my implementation for you to know, it might be to merge the two and thus work on more generic models, if you agree please inform me of how I can send you my project for you to know. Greetings!

juliogamasso, 2019-05-02:
[image]

lukevp, 2019-05-02:
Thanks for the additional info and for sending the screenshot over!  

Actually in this case I believe your print driver may not be set up for raw printing, and it doesn't have to do with the image command itself.  

The reason I think that is at the beginning of == [ Start Full Receipt ] ==. 
This is how it is supposed to appear:
[image]

 Before that == [ ... on your example, you can see the characters VB@a- or something like that.  There should be nothing before the ==, since that is an initialize / start monitoring / line feed command which are non-printable.  

The image commands start with [GS] ( L.  [GS] is a special character.  You can see after the == [ Start Full Receipt ] == that a (L is printing in your example.  So I am pretty sure it is due to the way the commands are being sent to the printer.

How is the printer connected (USB?), and what driver is it using?  Are you printing from Windows?

You can post the code example to github and link me to the repository, or you can just upload it here as a zip file if you'd like.  I'd like to see how the other library communicates with the printer, maybe it does something to tell the printer to use ESC/POS or to switch to raw mode.

juliogamasso, 2019-05-02:
Hello @lukevp, create a repository with the name ESCPOS_Printer, please check if access is verified. In the test with your project I used COM port and network

lukevp, 2019-05-04:
@juliogamasso Where did you create the repository?  I can check it out if you can send me a link.

The only difference I can think is the way the connection is set up or the bytes are sent across, or maybe some special initialization, since it's just plain ESC/POS otherwise.

juliogamasso, 2019-05-04:
https://github.com/juliogamasso/ESCPOS_Printer
I sent a developer invitation to you.

juliogamasso, 2019-05-04:
https://github.com/juliogamasso/Teste
I sent a developer invitation to you and moved the repository to public.

montyclt, 2019-05-04:
@lukevp I tested it and I get this print:

[image]
[image]

lukevp, 2019-05-04:
@montyclt  thank you for testing ! What type of printer was this for, and what communication method?

montyclt, 2019-05-04:
> @montyclt thank you for testing ! What type of printer was this for, and what communication method?

VivaPos P83 printer connected via ethernet.

lukevp, 2019-05-04:
@juliogamasso I tried to use your printer library but nothing printed out, so I modified it so that I could capture the byte stream output from the program.  I then placed another test case in my tester program, so it will use all of the COM and file/networking logic but with the byte array from your program.  

Please give this a try at your earliest convenience, and see if option 7 prints the XING-SAMPLE image or not.   You will have to clone the project and switch to branch bugfix/3-image-support which contains this test case.

If it prints properly, then I will compare the two generated byte arrays between my version and yours and determine where the issue is.  If it doesn't print properly, then the problem is not in the ESC/POS command generation, but somewhere else.

FYI, this does print properly on my Epson printer still.

lukevp, 2019-05-04:
@montyclt  if you don't mind, could you try the same branch bugfix/3-image-support and try option 7 to see if it prints on your printer?

montyclt, 2019-05-05:
[image]

juliogamasso, 2019-05-06:
@lukevp 
I tested the printers I have, they all printed correctly. The Epson and Tanca printer use Serial connection, Bematech used network. If I can help with anything else please let me know.

[image]

lukevp, 2019-05-07:
Thanks @juliogamasso @montyclt for testing!  I'm glad both of these printed, that means the issue is just in the image byte command generation and should be straightforward to fix.  I may not be able to work on this until this weekend, but I will have a fix for you as soon as I am able to.

lukevp, 2019-05-11:
@juliogamasso  @montyclt I have identified the issue.  The old library from Juliogamasso appears to only support the GS v 0 command for printing bit images, and not the newer GS 8 command.  Epson recommends not using the GS v 0 command for new libraries because it is legacy and may not be supported by future printers, but since it appears that it is common in non-epson printers for the new GS 8 command to be unsupported, I have added support for GS v 0 to the library!

Please check out branch bugfix/3-image-support, and try the test #7 - legacy image support.  It works properly on my Epson (although it does cause horizontal striping on the largest images).  If it works on the Bematechs, I will complete this issue and release this updated version on Nuget.  I am also opening another ticket to add dithering support to the image conversion, so it will not only support black and white.

Thanks in advance for all your assistance!

montyclt, 2019-05-12:
> @juliogamasso @montyclt I have identified the issue. The old library from Juliogamasso appears to only support the GS v 0 command for printing bit images, and not the newer GS 8 command. Epson recommends not using the GS v 0 command for new libraries because it is legacy and may not be supported by future printers, but since it appears that it is common in non-epson printers for the new GS 8 command to be unsupported, I have added support for GS v 0 to the library!
> 
> Please check out branch bugfix/3-image-support, and try the test 7 - legacy image support. It works properly on my Epson (although it does cause horizontal striping on the largest images). If it works on the Bematechs, I will complete this issue and release this updated version on Nuget. I am also opening another ticket to add dithering support to the image conversion, so it will not only support black and white.
> 
> Thanks in advance for all your assistance!

[image]

[image]

juliogamasso, 2019-05-13:
I tested the printers I have, everything is working. Some newer ones do not accept the command GS v 0, but as we have the GS 8 meets both new and old. Nice implementation @lukevp! Thanks!

lukevp, 2019-06-04:
Closed - this issue has been resolved and merged.  Thanks everyone for testing!

closed 2019-06-04 by lukevp