lukevp/ESC-POS-.NET #8: add dithering support for images

lukevp, 2019-05-11:
Add support for dithering algorithms to image printing library to enable higher quality printing of non-black-and-white images.

lukevp, 2019-05-19:
@juliogamasso @montyclt feature/8-add-dithering-images has been created, which prints images with much higher fidelity by using a dithering algorithm.  However, on my Epson it also has corruption in the final image (due to a buffer overflow).  Could you give this a try and see if the image printing of the kitten looks better now, and if you see corruption in any images?

montyclt, 2019-05-20:
Ok, I will try it this night (utc+2)

juliogamasso, 2019-05-20:
@lukevp  I did the test, the printing came out very good, but in test 6 using Epson TM88V an error occurred 'Insufficient buffer for the data requested'

montyclt, 2019-05-22:
Should I use image test o legacy image test?

I executed image test and I get strange characters (because my printer requires legacy images), then I get a print with info of my printer (the same info if I press power + feed button in the printer), then the paper is cut and I get printer info again, then the paper is again cut and I get strange characters, low quality image and strange characters, then an exception in throwed and program ends.

After doing this test, I cannot connect to the printer again, even turn off and turn on again. Possible data ramain into the buffer?

[image]

lukevp, 2019-06-04:
@montyclt  were you able to get your printer to connect again?  Sorry for the trouble!  You could modify the test to print using legacy, but it might be better for you to wait and retest after I fix the large buffer bug on in issue #10

montyclt, 2019-06-05:
> @montyclt were you able to get your printer to connect again? Sorry for the trouble! You could modify the test to print using legacy, but it might be better for you to wait and retest after I fix the large buffer bug on in issue #10

Yes, I done a factory reset in printer and I can reconnect again. Maybe a data stay in the buffer for all time? I don't know why happened, but I solved. Thank you!

juliogamasso, 2019-06-05:
I also had problem with buffer size but not to the point of not being able to connect. I made a byte extension to divide according to the size of the printer buffer and it is working.

montyclt, 2019-06-06:
> I also had problem with buffer size but not to the point of not being able to connect. I made a byte extension to divide according to the size of the printer buffer and it is working.

Thanks for the info, if you can, document it to avoid the problem in other libraries. Thank you!

I'll follow #10

lukevp, 2019-10-03:
@montyclt  @juliogamasso The dithering and large byte array issues should all be working properly now, check out PR #30 and let me know!

montyclt, 2019-10-04:
> @montyclt @juliogamasso The dithering and large byte array issues should all be working properly now, check out PR #30 and let me know!

Thanks @lukevp. Can you document the problem about large byte array? I will execute all test this weekend and inform you about the result :)

lukevp, 2019-10-05:
@montyclt on release/1.0 there is a test case called Large Byte Arrays.  This passes a stacked image array to the serial printer in one shot, so it tests the buffering of the underlying stream.  This test failed with the old implementation using SerialPortStream, but on Windows 10 over Serial it prints perfectly.  I updated the PR just now to use the legacy print command that your printer supports, so make sure you get the latest version before you test.

montyclt, 2019-10-07:
I tested large byte arrays with network printer and works ok. I get printed 20 images, 10 kitten and 10 companion cube.

Off-topic: A Portal fan here! 😄

lukevp, 2019-10-08:
That's great, so glad that it is fixed! I'm also a portal fan although I haven't gotten a chance to play in a while!

lukevp, 2019-12-26:
Dithering is supported in 1.0.0 release.

closed 2019-12-26 by lukevp