lukevp/ESC-POS-.NET #10: Corrupt data writes when byte size too large

lukevp, 2019-05-14:
When writing a really large byte array (15+ images in a single call, for example), C# will error out and the byte streaming will fail, or it will write corrupt data.

juliogamasso, 2019-10-01:
Good morning @lukevp , did you analyze anything about this problem? I am also having trouble printing with Epson TM88V when the array is too large. I tried to segment the array not to be sent at once but it didn't solve. Thanks!

lukevp, 2019-10-01:
Julio,

I have a branch open that rewrites the serial implementation to use
system.io.ports in .net core 3, give it a try and see if it works for you!
It solved the issue for me, but if you turn on live status back monitoring
it does not currently work, that is why I haven't yet merged it.

juliogamasso, 2019-10-01:
Is this branch bugfix-10-corrupt-writes-on-large-buffers? It's not compiling here

[image]

lukevp, 2019-10-01:
yeah that's the branch.  I will have to wrap the stream calls in an API that calls the system.io.ports internally since reads on the system.io.ports are blocking, so you can only perform the reads if there is data available.   For testing purposes on your large byte array you could comment out some of that Read method, it is only used for the live status back right now. It's not necessary for printing.

juliogamasso, 2019-10-01:
I tried to test but to no avail, it prints nothing, Thanks.

lukevp, 2019-10-03:
@juliogamasso  Try out release/1.0 and see if it works for you.  I fixed all the issues with reading and writing using system.io.ports.  Also this is a breaking change because system.io.ports only lets you use COM ports below COM8 on Windows, so that could also be part of your problem.

lukevp, 2019-10-08:
@juliogamasso hope you get a chance to test this soon, I am planning on releasing version 1.0 officially this weekend!

juliogamasso, 2019-10-08:
@lukevp 
Tested with the Epson TM-T20

I did the 8 tests and they were all successfully printed.
My problem was a very large single image was crashing the printer. I added a method to split the image into smaller parts, I managed to solve it that way. thanks!

lukevp, 2019-10-08:
Which part was crashing, the image resize part?

juliogamasso, 2019-10-08:
In the tests I had no problem, what I had was with an image that was large but not in the tests of your branch. I had only one continuous image and I was locking the printer

lukevp, 2019-12-26:
This is fixed in 1.0.0 release.

closed 2019-12-26 by lukevp