lukevp/ESC-POS-.NET #12: USB POS Printing

tibovds-odisee, 2019-05-16:
Hello,

 I might have overlooked it;

Is there a way to print when the printer is nor on the network, nor using a Serial Connection?
I currently have my thermal printer attached through USB, can I still use this library?

Thanks!

lukevp, 2019-05-19:
Most printer drivers support virtual COM ports, which work great with the library. I actually do the development primarily using an Epson TM-20II printer connected over USB using Virtual COM. It works on Linux as well. Let me know what printer you have and I may be able to help with some more specific suggestions.

tibovds-odisee, 2019-05-19:
Thanks for the reply, I have figured out how to use a virtual COM port for my printer. The library now works great with it. :)

lukevp, 2019-05-19:
Perfect! Glad it is working for you. Feel free to contribute or join in the planning or conversation around additional features. The library is also available via NuGet.

closed 2019-05-19 by lukevp

Prakash19921206, 2020-08-08:
Hi @lukevp ,
i'm testing a less popular thermal printer ([HOIN POS-80-Series](http://www.hoinprinter.com/en/products)) on windows 10
i don't think this printer driver doesn't supports Virtual COM
i created a Virtual COM port(COM2) as i had only 1 by default (COM1)
but couldn't map this usb printer to COM2.
referred [this ](https://superuser.com/questions/923426/how-to-map-a-virtual-com-port-to-a-physical-usb-port)post and tried below command
`NET USE COM2: \\DESKTOP-HJ34TQ9/POS_80_share /persistent:yes`
it says 

> System error 66 has occurred.
> The network resource type is not correct.

[old repo ](https://github.com/mtmsuhail/ESC-POS-USB-NET)works well with usb device (printer) name.
is there any work around for this?

Thanks & Regards
Prakash