lukevp/ESC-POS-.NET #13: Print pdf file

juliogamasso, 2019-06-03:
Hello @lukevp, did you get to read something about PDF printing using ESC POS? I could not figure out how to do it using the commands. Thank you!

montyclt, 2019-06-04:
Maybe transforming the PDF file into bitmap image and print it with image commands?

juliogamasso, 2019-06-04:
I also thought about this, but it is very difficult to convert pdf into image

lukevp, 2019-06-04:
@juliogamasso there's nothing in the ESCPOS standard that would let you print a PDF, you would have to use a separate library to rasterize it before printing.  What sort of PDF do you have that would be printable on a receipt printer?  Perhaps you could get the data in a different format?

juliogamasso, 2019-06-04:
The receipts can come in the format of the attached image, would it be possible to align a QrCode the way it is in the image? I have already implemented but only line by line, not side as it is in the image. If it is possible to mount an equal receipt using the commands we have already solved the problem.
[image]

lukevp, 2019-06-04:
To do this natively in ESCPOS, I believe this is something you would have
to use Page Mode for, which is not currently supported in the library. Page
Mode lets you control the printing position and place elements next to each
other. We don't need that for receipt prints since the area is so narrow,
so I haven't implemented Page Mode and don't plan on adding it, but it is
something you could add. Or you can try converting the PDF to a PNG and
printing that way.

juliogamasso, 2019-06-04:
Allright thanks.

closed 2019-06-04 by juliogamasso