lukevp/ESC-POS-.NET #14: Native USB POS printing

ThrowingSpoon, 2019-07-29:
Are you planning on supporting native USB printing? I am looking to print, and receive events back (such as low paper, failed to cut, failed to print. etc) via USB without using a virtual COM port. Having looked through this code it is very thorough and would be amazing if it also supported USB! 🤞

lukevp, 2019-07-29:
Liam,
The current virtual com port over USB implementation supports real time events so you can register callbacks for these states. The reason I chose to implement Virtual COM over native is that native USB required installing a device driver as well as registering a print queue on Windows, which is error-prone to do in an automated way and required the same amount of work as virtual com. Additionally, if you don't use the windows spooler then I didn't see a way to implement the printing in a generic way. What platforms are you targeting, and what do you see as advantages of native USB? I am happy to review and merge a PR and assist you with the USB support if it seems like it would be helpful. the esc/pos command generation and the communication protocols are already decoupled so it shouldn't be too hard to add support.

ThrowingSpoon, 2019-07-29:
Luke,
Thanks for clarifying that, now you mention it, virtual COM over USB does sound like a much less stressful option! I will use my printer with a virtual COM port over USB and work with callbacks.
The main reason I wanted to use native USB was to reduce config overhead and increase speed available for firmware changes etc... But after considering your points, it does seem native USB is not the way to go,
Thanks,
Liam.

lukevp, 2019-07-29:
Great! Closing this issue but please let me know what you think once you start using it, or if you have any troubles with the library! There are some common functions still missing but it's getting pretty solid.  I would appreciate some help writing a getting started guide or fleshing out the readme more.

lukevp, 2019-07-29:
I should mention that l, while I haven't done a firmware update yet, you can use very high baud rates over virtual com, and the library is tuned for this use case so it can stream data about as fast as my Epson printer can print. This library is much faster than a lot of implementations I've seen, it does all writes asynchronously in the background and intelligently flushes the buffer as well.

closed 2019-07-29 by lukevp

ThrowingSpoon, 2019-07-29:
Sounds good! If I get some snippets together from when I use the library I'll send you them or collab to do a getting started guide

ThrowingSpoon, 2019-07-30:
Luke,
Have you got any snippets for registering and handling printer events using your library please? I am trying to simulate a paper low event and have it call a callback that just logs to the screen.
Thanks,
Liam.

ThrowingSpoon, 2019-07-30:
I did try setting up a dev project similar in the way you did your tests, and noticed that in the base printer, the `var b = _reader.ReadByte();` never reads, even if I take out the paper or try to cause any event

lukevp, 2019-07-30:
You should try running the full desktop tester, it has status monitoring built in with an event handler already hooked up to write the status flags to the console. You have to enable automatic status back before you will get live responses. I have only tested with an epson tm-t20ii but I get events for paper drawer open/closed, out of paper, and feed button pressed.

ThrowingSpoon, 2019-07-30:
I did try that and cannot seem to get any status' back, I am using a TG2460H III thermal printer by a company called Custom, I could not find any automatic status option in any of the configuration software from them, I have tried configuring the printer as Custom POS and a TGH Emulations, could this be a config or just a compatibility issue?

lukevp, 2019-07-30:
Your printer appears to not support automatic status back, so you would have to build out the support for regular status messages. I only built in automatic status back because it is the most modern Epson status technique since you can know immediately if there is an issue. With the older status commands you have to poll rather than the printer being able to tell you. Would you like to open a new issue for implementing regular status back, and I can help you with any questions you have? The library is pretty straightforward to add new commands to.

lukevp, 2019-07-30:
Regarding compatibility, if the settings are incorrect on the printer, some commands like text printing will still work, but you should be able to also print images and barcodes. If that is not working, then it is likely not receiving raw print commands.

ThrowingSpoon, 2019-07-30:
> Your printer appears to not support automatic status back, so you would have to build out the support for regular status messages. I only built in automatic status back because it is the most modern [...]

Thanks, I'll open that now

ThrowingSpoon, 2019-07-30:
> Regarding compatibility, if the settings are incorrect on the printer, some commands like text printing will still work, but you should be able to also print images and barcodes. If that is not working, then it is likely not receiving raw print commands.

Yes I can print barcodes, text, etc... It is just receiving status' from the printer that appears to be the issue

ThrowingSpoon, 2019-07-30:
Having checked the command bytes, it appears my printer has a similar, but not exact same command set as the Espon printers, I will see if substituting commands will work, as I have hacked together some bits that emit events when the feed button is pressed