lukevp/ESC-POS-.NET #15: Implementing polling for printer status

ThrowingSpoon, 2019-07-30:
For printers that do not automatically return status', could we implement a method for polling printer status'? currently the printer status for my TG2460H III is always null as no automatic data is received.

lukevp, 2019-08-22:
Hey @throwingspoon, any luck adding the status functionality?

ThrowingSpoon, 2019-08-22:
Hey @lukevp it turns out the Custom TG printers had a slightly different implementation for their commands than the Epson printers did, and changing the command to the correct sequence of bytes got it working

lukevp, 2019-08-23:
That's great? We can make another emitter instead of the Epson one. What is the sequence for yours? Is the response message the same?

lukevp, 2019-08-24:
Hey @ThrowingSpoon  github had a mention outage yesterday so wanted to tag you in case you didn't see this.  Do you have a fork where you added the status bytes?

ThrowingSpoon, 2019-08-27:
> That's great? We can make another emitter instead of the Epson one. What is the sequence for yours? Is the response message the same?

The response was not the same, however I did make a class to hold the response in, but it was for a custom printer, maybe a better pattern for the commands/responses would be to have a base class with only very basic commands there, and have each implementation extend from that?

lukevp, 2019-09-14:
@ThrowingSpoon  could you provide your changes as a PR or link me to them so I can check them out? I only have a single, modern Epson printer to test on, so the library hasn't been extended a whole lot for compatibility.

lukevp, 2019-12-26:
No response, closing issue.  Please feel free to comment and we can reopen if you'd like to explore adding this to the library! Thanks.

closed 2019-12-26 by lukevp