lukevp/ESC-POS-.NET #16: Dispose pattern not implemented properly

darkurse, 2019-08-12:
Hi there,

first of, thank you very much for your project !!

Regarding the Dispose pattern. NetworkPrinter doesn't implement a safe disposable pattern.

When using the simple following code, things go kaboum

```C#
using (var p = new NetworkPrinter())
{
   // [....]
}
```

That's because you also call Dispose() in the finalizer without checking if resource has already been disposed.

lukevp, 2019-08-12:
Good points and I'm glad you are finding the library useful!
To be honest I'm not sure why they don't implement dispose properly. Is this something youd be willing to test and make a PR for?  We should fix this in all of the printers since they all have that same issue. 
How about using the null conditional operator for the close calls?

darkurse, 2019-08-13:
Hi Luke, I had to deliver this fix quickly so I actually forked your project to do so and put the nuget package on private feed.

However I can push you a PR later on. FYI, that's my fix

```C#
public void Dispose()
{
    if (_socket != null)
    {
        _writer?.Close();
        _reader?.Close();
        _socket.Close();
        _socket = null;

        GC.SuppressFinalize(this);
    }
}
```

1. Even if Dispose is called twice manually, it won't attempt to release the socket twice.
2. Writer and Reader are Null Reference protected
3. If dispose was called manually, there is no longer need to call the finalizer.

lukevp, 2019-10-03:
@darkurse  this should be resolved in a generic way in PR #30, please check it and comment!

lukevp, 2019-12-26:
Resolved in 1.0.0 release.

closed 2019-12-26 by lukevp