lukevp/ESC-POS-.NET #17: Support for Display

dirkolene, 2019-08-14:
Hello, 
I would like to know if it is possible to use the customers display commands. If so, how ?

YZahringer, 2019-08-15:
ESC/POS Commands for Customer Display: https://reference.epson-biz.com/modules/ref_escpos_dm_d_en/index.php?content_id=47#commands

ESC/POS Command to Drawer Kick out: https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=195

lukevp, 2019-08-21:
@dirkolene @YZahringer  do you have a line display and a cash drawer? I don't have either, so I didn't implement either of these calls.  I'd be happy to add support to them as long as someone can validate that they work!

lukevp, 2019-08-31:
@dirkolene  @YZahringer  pinging for an update.  If you have a line display or cash drawer and can test these commands, I am happy to add them!

YZahringer, 2019-08-31:
@lukevp Sorry for the late answer. Yes I have a Line-Display Epson DM-D110 and a Cash Drawer for tests. I would be happy to validate the implementation 🙂

lukevp, 2019-08-31:
@YZahringer  Great!  

Do these ESC/POS commands pass through the printer, or are they sent directly to the line display? Would you be running 2 instances of the SerialPrinter class at once, one for the display and one for the printer?  I haven't tested the library in this scenario so it would be good to know if you're doing that within the same application and how it works out.

Do you also happen to have a buzzer?

The cash drawer should be straightforward to add so I'm more curious about the display.

How would you want to interact with the display from the library? would you just want to issue the raw commands with command emitters, or would you want to have a higher level interface of some kind?

For example, I see that there's ways to set scroll modes and let the line display scroll the screen.  However I would prefer to just repaint the screen every time a call is made, and implement scrolling and animation timing on the C# side.  That would give more flexibility, the only downside would be if redrawing the whole screen was laggy or glitchy looking.   If it's not then there's no need for scrollers, blinkers, window/page modes, etc. of the display as that can easily be implemented in code.

What is the required subset of commands in order to use the line display? If you don't need scroll modes, it seems like the default would suffice, so you would just need to specify to the library how many lines and columns you have, and then all you really need are:
ESC @ - initialize (already implemented)
CLR - clear screen

And then writing to the screen would just be the existing print commands, and we could create a wrapper that would help you not exceed your character boundaries.

And optionally you might want to use style modes like
US r for reversing the text
or possibly the Kanji or character set definitions.

Some config commands might be helpful like
US X for brightness

All the windowing and scroll modes do not seem necessary, similarly to how the library does not include support for Page Mode or obsolete image print commands.  I'm trying to keep the command set focused on what is really useful and implement the rest on the C# side.

YZahringer, 2019-09-04:
Usually we use two COM ports, one for the printer and one for the display. The cash drawer connected to the printer. Controlling the display through the printer would be possible with ["ESC ="](https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=191) command. This could be interesting but not a priority for us.

We manage COM/USB devices from a cross-platform application (UWP/Droid/iOS) with Xamarin. To access the COM port we use `System.IO.Ports` which is supported by .NET Core on Windows and Linux, version 3.0 brings a lot of improvements. 

We have not yet integrated your library, but we have tested with a class that inherits `BasePrinter` and calls directly `System.IO.Ports.SerialPort` without problem.

We do not currently use the Buzzer but it could be useful.

For the display, I think it would be useful to implement ["ESC @"](https://reference.epson-biz.com/modules/ref_escpos_dm_d_en/index.php?content_id=37) to init, ["ESC v"](https://reference.epson-biz.com/modules/ref_escpos_dm_d_en/index.php?content_id=39) to check status/handshake and ["CLR"](https://reference.epson-biz.com/modules/ref_escpos_dm_d_en/index.php?content_id=2) to clear display.

Configuration can generally be done through physical switches on the display but would be interesting to do it from code.

lukevp, 2019-09-04:
Could you open a PR with the system.io.ports move? I didn't use it for the original implementation because it was not cross platform but I plan to move to it in the next release when 3.0 is production. It turns out that all the issues in #10 were related to the serialportstream library. However, I'm having issues with the read, the previous port library allowed a blocking read call independent of the writes. In system.io.ports as soon as you enable status monitoring and it starts the read thread, it blocks the writing.

YZahringer, 2019-09-04:
We haven't done any advanced tests, but the status is updated correctly.

I will do a PR when the 3.0 will be released, after the .NET Conf on September 23rd. 

Here is the example code that is currently used with .NET Core 2.2:
````cs
    public class StandardSerialPortPrinter : BasePrinter, IDisposable
    {
        private readonly SerialPort _serialPort;

        public StandardSerialPortPrinter(string portName, int baudRate) : base()
        {
            _serialPort = new SerialPort(portName, baudRate);
            _serialPort.Open();

            _writer = new BinaryWriter(_serialPort.BaseStream);
            _reader = new BinaryReader(_serialPort.BaseStream);
        }

        ~StandardSerialPortPrinter()
        {
            Dispose();
        }

        public void Dispose()
        {
            _writer.Close();
            _reader.Close();
            _serialPort.Close();
        }
    }
````

lukevp, 2019-12-26:
@YZahringer @dirkolene  I am going to leave this open for now in case either of you are interested in adding support for the customer display or testing what I have.  I had what I thought was a working customer display but it did not have drivers compatible with current Windows releases, so I'm not able to proceed on testing this one.  

I have added the CLR command as BaseCommandEmitter.Clear in release/1.0.0 that will be pushed to Nuget shortly, that should be the only command other than initialize that you need (assuming automatic status back works or you do not need to read status).

YZahringer, 2020-01-29:
Cash-drawer support is in #34 🎉

@lukevp Do you have a release date for 1.0.0? or new pre-release?

lukevp, 2020-02-11:
@YZahringer everything is merged to master, including the display and cash drawer support!  Check out  the latest RC at https://www.nuget.org/packages/ESCPOS_NET/1.0.0.1-beta2

It's a prerelease so you will have to enable prerelease packages to install it.  Let me know if you have any issues and we can cleanup before the official release, but all our testing so far has been good.

lukevp, 2020-02-11:
@dirkolene  pinging you since you opened this issue, the latest RC above has support for customer displays and cash drawers. Please let me know if you have any issues!

lukevp, 2020-02-19:
No feedback, closing issue.  Promoting prerelease package to release package version 1.0.0 on Nuget!

closed 2020-02-19 by lukevp