lukevp/ESC-POS-.NET #18: QRCODE Implementation

netgg93, 2019-08-15:
Hey Guys

I have implemented Qr-code printing functionality. I'm a beginner in programming so I'm not sure how to submit the code here. Is there a way way I could submit my code here for you guys to review. 

Cheers!!

[image]

netgg93, 2019-08-15:
I have created pull request for review of the code.
Cheers!!

lukevp, 2019-08-20:
Hey @netgg93 , thank you for contributing! The code looks good, although I haven't reviewed it in depth.  The only issue with proceeding on the PRs is that your changes are in separate branches, so it's difficult to review and comment on them like this.  We're using GitFlow for this project, so it would be more appropriate for you to create a PR for merging your remote develop branch into the project's develop branch.

So next steps for you would be to close all 4 of your PRs, then merge your different branches into your develop branch, and then create a single PR for netgg93/develop to lukevp/develop that includes all your changes.  I'll then review and comment on it and once we resolve any issues it will get merged.

Thank you again for your work!

lukevp, 2019-08-24:
hey @netgg93 please see above. I'm closing this as a duplicate of #4 and I am closing the PRs, but if you follow the directions above and tie your PR to issue #4 we can review it there. Thanks again!

closed 2019-08-24 by lukevp