lukevp/ESC-POS-.NET #24: Code Page?

asafgo, 2019-09-04:
Hi,
How to set code page?
For example Windows-1255 for Hebrew characters (POS code page set to 26)

Thanks,
Asaf

lukevp, 2019-09-05:
@asafgo switching code pages is not yet implemented, but it's pretty straightforward to make changes to the library. Why don't you fork and give it a try, and send us a PR if you add it in? Would be a great feature to have!

asafgo, 2019-09-05:
Hi,
In what class code page is so I will happy to give it a try as I am not professional programmer?

asafgo, 2019-09-05:
Hi,
I have succeeded to print in Hebrew, thanks @lukevp 

`    public static partial class Tests
    {
        public static byte[] Printing(ICommandEmitter e) =>
         ByteSplicer.Combine(
                e.Print("Multiline Test: Windows...\r\nOSX...\rUnix...\n"),
                e.FeedLines(1),
                e.RightAlign(),
                System.Text.CodePagesEncodingProvider.Instance.GetEncoding(1255).GetBytes(Reverse("הדפסה בעברית.")),  //Needs to install "System.Text.Encoding.CodePages" https://www.nuget.org/packages/System.Text.Encoding.CodePages/
                e.FeedLines(2),
                e.LeftAlign(),
                // TODO: sanitize test.
                e.PrintLine("Feeding 250 dots."),
                e.FeedDots(250),
                e.PrintLine("Feeding 3 lines."),
                e.FeedLines(3),
                e.PrintLine("Done Feeding."),
                e.PrintLine("Reverse Feeding 6 lines."),
                e.FeedLinesReverse(6),
                e.PrintLine("Done Reverse Feeding.")
             );

        public static string Reverse(string s)
        {
            char[] charArray = s.ToCharArray();
            Array.Reverse(charArray);
            return new string(charArray);
        }
    }
`

closed 2019-09-05 by asafgo