lukevp/ESC-POS-.NET #25: Can't print Image

asafgo, 2019-09-05:
Hi,
When printing image the results is like in the picture.
[image]

lukevp, 2019-09-05:
@asafgo thanks for trying the library! Can you specify what printer you're using, how it's connected, and what driver? This looks like an issue w/ a driver not passing raw commands through.

asafgo, 2019-09-05:
Hi,
My printer is [http://www.zywell.net/product/ZY606-Thermal-Receipt-Printer.html](url)

lukevp, 2019-09-05:
Can you try using branch feature/8-dithering-images and try a legacy print? There are 2 different print commands and non-epson printers do not always implement the latest command set for ESC/POS.

asafgo, 2019-09-05:
I am sorry but I didn't understood what should I do?
What are the 2 different print commands please? so I will ask the manufacturer of the printer.

lukevp, 2019-09-05:
Check out the branch feature/8-dithering-image instead of master. Then run EscPos.ConsoleTest and choose "legacy images" as the test. See if that works!

asafgo, 2019-09-05:
Yes! Legacy Print works perfect.
[image]

lukevp, 2019-09-14:
Hey @asafgo  thanks for the follow up, glad the legacy print works! I will add an enhancement request to make different emitters, and for non-EPSON emitters it will use the legacy print technique by default.

closed 2019-09-14 by lukevp

MitkoMitrov, 2024-12-03:
Hey @lukevp I cannot find the branch that You mentioned before. Could You help me, I got the same error as @asafgo. Thanks in advance.