lukevp/ESC-POS-.NET #26: Replace external libraries by .NET standard libraries

Xenoage, 2019-09-05:
Both the .NET Framework and .NET Core now contain methods for image processing and serial port connection.
I replaced `SixLabors.ImageSharp` by `System.Drawing.Common` and `SerialPortStream` by `System.IO.Ports`.
If you are interested to get rid of the external dependencies, have a look at my commits on https://github.com/Xenoage/ESC-POS-.NET

lukevp, 2019-09-05:
@xenoage can you send a PR for your changes? The problem I have with system.io.ports is that when live status back monitoring is turned on, the read thread was blocking the writing after the first call. Another issue to keep in mind is that system.io.ports was not cross platform when I started the development, and it only became cross platform w/ .net core 3.0 runtime, which is not a stable release for another few weeks. That's why I initially used serial port steam library.

As for system.drawing.common, I have read a lot of advice to avoid system.drawing based on performance issues, and it also appears to be windows only. Is it cross platform in .net core 3?

Xenoage, 2019-09-05:
@lukevp I didn't know about the `SerialPort` issue, but had no problems so far with it in my projects (running with .NET Core 3 both on Windows and Linux devices). `System.Drawing.Common` in .NET Core 3 works perfectly on Linux, too.

I have added another library, `System.Text.Encoding.CodePages` to support the encoding of the text to `IBM00858`. Using this encoding, e.g. German Umlaute (a, ö, ü, ...) are displayed correctly (see https://stackoverflow.com/a/10640118/518491 )

I'll create a PR for my changes.

closed 2019-09-05 by Xenoage

lukevp, 2019-09-05:
Great! Thank you so much @xenoage, I will check this PR out over the weekend!