lukevp/ESC-POS-.NET #31: Printing code128 barcode

toumir, 2019-10-26:
Hi, 
can anyone tell me how to print **code128** barcode ? always I get **blank** paper!
Thanks in advance

ThrowingSpoon, 2019-10-26:
Have you tried ensuring your paper is flipped the correct way? As thermal paper typically only can be printed on one side, can you print normal text or use the printers 'test' button?

toumir, 2019-10-26:
Hi, 
thank you Sir for the quick reply, 
Yes I can Print QrCode ,EAN13, ITF, CODE39, EAN8, UPC_A, UPC_E,  **Only the Code128 codebare not working for me!**
Can you Help?
thanks in advance

toumir, 2019-10-26:
FYI : I send three codes to the printer: CODE39, CODE128,CODE39, always I get a blank line, see the attached image
[image]

lukevp, 2019-10-26:
@toumir This is a bug in the CODE128 support that I've just pushed a fix for.  Check out the latest release candidate on release/1.0 branch and let me know if test case #4 works for you! This should print all valid code types that are defined in the ESC/POS standard.

ThrowingSpoon, 2019-10-26:
Are you sure your printer supports this encoding? Try to check any printer documentation you can find, what printer are you using?

Are you able to capture the response from the printer when you send it that barcode that does not print?

[20191026_210243]<https://user-images.githubusercontent.com/10366325/67625382-e0d10900-f83d-11e9-8c8c-e40efd0c892e.png>

lukevp, 2019-10-26:
[image]
This is what the new test case looks like when printed.

toumir, 2019-10-26:
Hi @lukevp @ThrowingSpoon Thanks for your help and for your time, 
Problem Solved :), some code not printed because my printer not support those encoding (Self-Test attached)
Thank you 
[image]
[image]

lukevp, 2019-10-26:
That's great! I will include these fixes in the 1.0 release on Nuget as
soon as I am done preparing it!

closed 2019-10-26 by lukevp