lukevp/ESC-POS-.NET #33: Access Denied to COM port

kodejack, 2019-12-10:
When using the serial printer with UWP, it throws an exception with Access Denied: COM#

ThrowingSpoon, 2019-12-10:
If you are on windows, check Device manager to see if your COM port is visible, then check that the COM# number lines up with that error, if they are not the same, try changing the COM port in your code to the correct port, if they are the same, then check that another process has not opened that COM port and is hogging it.

kodejack, 2019-12-10:
Thank you for you're swift reply. I have now moved it from COM1 to COM3, still the same result. It's a vanilla install of Windows 10 IOT, connecting to an Epson TM88V printer. I can see the COM ports and the configuration seems correct. I have also added the serial port capabilities to the manifest

ThrowingSpoon, 2019-12-10:
Are you able to get a stack trace and copy and paste it here? Might give some more info as to where it is failing

lukevp, 2019-12-11:
@kodejack  Is this with the master branch?  Could you try it with the release/1.0 branch? It uses the microsoft serial port implementation instead of the cross-platform one that the current master branch does.

kodejack, 2019-12-11:
Here is the stacktrace
```
   at RJCP.IO.Ports.Native.WinNativeSerial.WinIOError()
   at RJCP.IO.Ports.Native.WinNativeSerial.Open()
   at RJCP.IO.Ports.SerialPortStream.Open(Boolean setCommState)
   at RJCP.IO.Ports.SerialPortStream.Open()
   at Zonal.Shared.Plugin.OnDemandPrinter.ESCpos.Printers.SerialPrinter..ctor(String portName, Int32 baudRate)
   at Zonal.Shared.Plugin.OnDemandPrinter.SerialOnDemandPrinter.<>c__DisplayClass1_0.<InitialisePrinter>b__0()
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
  ```

kodejack, 2019-12-11:
Yes, it is the master branch, I need it to work cross-platform for the IP printers, the serial printer is just UWP. Will v1.0 no longer be cross-platform?

kodejack, 2019-12-11:
Here is the ConsoleTest app stack trace and some screen shots

```
ESCPOS_NET Test Application...
1 ) Test Serial Port
2 ) Test Network Printer
Choice: 1
COM Port (eg. COM5): COM1
Baud Rate (eg. 115200): 9600
Unhandled exception. System.ArgumentException: The given port name (COM1) does not resolve to a valid serial port. (Parameter 'portName')
   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   at System.IO.Ports.SerialPort.Open()
   at ESCPOS_NET.SerialPrinter..ctor(String portName, Int32 baudRate) in C:\Projects\Zonal\Temp\ESC-POS-.NET\ESCPOS_NET\Printers\SerialPrinter.cs:line 15
   at ESCPOS_NET.ConsoleTest.Program.Main(String[] args) in C:\Projects\Zonal\Temp\ESC-POS-.NET\ESCPOS_NET.ConsoleTest\Program.cs:line 52
```

[image]
[image]

lukevp, 2019-12-11:
Can you share the source of your app, or make a minimal reproduction repo and share it? 

Do you have the serial port capability for your app?

https://github.com/microsoft/Windows-universal-samples/blob/master/Samples/CustomSerialDeviceAccess/cs/Package.appxmanifest

System.io.ports is cross platform as well but requires .net core 3, at the time the library was started there was no native cross platform serial port library.

kodejack, 2019-12-11:
Yes, I have the capabilities set. I've made a bit of a breakthrough. I've uninstalled the Epson drivers and set it up with NCR Config tool and I can now get the ESCPOS_NET.ConsoleTest app to work with the serial printer. Why the Epson utilities and drivers would cause this to block the port, I don't know. I'm going to pull the 1.0.0-beta into my plugin and see if I can get it working.

lukevp, 2019-12-26:
That's great @kodejack  glad it worked out.  I'm closing this issue and will place it in milestone 1.0.0 since it is fixed with the new version.

closed 2019-12-26 by lukevp