lukevp/ESC-POS-.NET #35: Thread.Abort raise PlatformNotSupportedException

oachkatzalschwoaf, 2020-01-03:
in the Dispose-method of BasePrinter
 `_readThread?.Abort();` throws `PlatformNotSupportedException` when the target-framework set to .NET Core  ( e.g. your testproject; ESCPOS_NET.ConsoleTest )

see: [https://docs.microsoft.com/en-us/dotnet/api/system.threading.thread.abort](https://docs.microsoft.com/en-us/dotnet/api/system.threading.thread.abort )

lukevp, 2020-01-18:
hey @kodejack is this fixed by your PR?  @oachkatzalschwoaf  you could check open PR #37 to see if it is fixed, this seems related to what kodejack was addressing.

oachkatzalschwoaf, 2020-01-19:
> hey @kodejack is this fixed by your PR? @oachkatzalschwoaf you could check open PR #37 to see if it is fixed, this seems related to what kodejack was addressing.

Yes he fixed this issue. => I've done a quick test with my Network printer (just dispose each time after printing then reconnect).

kodejack, 2020-01-22:
Great! Glad it's sorted.

oachkatzalschwoaf, 2020-02-10:
Fixed in Master

closed 2020-02-10 by oachkatzalschwoaf