lukevp/ESC-POS-.NET #36: TCP-connection/Timout

oachkatzalschwoaf, 2020-01-04:
When connecting over TCP, the printer closes the connection after 90 seconds inactivity with the reset flag. (Observed with Wireshark on my EPSON TM20II i use for my tests).

this cause that in BasePrinter -> Read() -> `ReadByte(); ` throws an System.IO,IOException each 100ms (as you catch the exception with `Thread.Sleep(100)`)
also Write() will throw the same exception.

here some logic is needed to check if the connection is lost. 
I have done some tests in my forked branch _bug/connection_ :
e.g. the connection can be checked as described here: [https://stackoverflow.com/a/14925438/12648992 ](https://stackoverflow.com/a/14925438/12648992 )
 with a method for the network-Printer e.g.:

```
public override bool IsConnected()
        {
            return !((_socket.Poll(1000, SelectMode.SelectRead) && (_socket.Available == 0)) || !_socket.Connected);
        }
```
also the `System.IO.IOException` exception of BasePrinters `Read()` might be handled as lost connection.

however with my changes i can only handle the dropped connection by the printer.
handle issues with the network like unplug eth-cable on the printer or restart raises additional issues (TCP-connection stayes open while the printer is gone)

lukevp, 2020-01-18:
@kodejack could we integrate this in your PR #37  ?  I don't have a TCP/IP printer (I use virtual COM over USB) so I'm unable to test.

kodejack, 2020-01-21:
I can have can give it a try, what is the expected behaviour? Should we attempt to reconnect if it times out?

oachkatzalschwoaf, 2020-01-21:
This is the current behavior with **your** unchanged ConsoleTest from `feature/remove-thread-abort` branch:
My printer is 192.168.1.190
[image]

First Lines: the connection with the printer is started (without Status Back Monitoring)
Then i send the data(i used 11: Cash Drawer)

after 90 seconds, the last red line: without additional data sent to the printer, the Printer will close the connection due to timeout(i think 90 sec is default for most Epson printer).

after the timeout if you try to send additional data(any print command from console), the write function will throw an IOException
==> currently not surrounded by try/catch

 _basePrinter.cs Line 81:_ 
`     Writer.Write(bytes, bytePointer, count);`

if you enable status-monitoring, in the read()-function you will see the repetitive IOException.  As mentioned here: https://github.com/lukevp/ESC-POS-.NET/pull/37#discussion_r368292154

my changes mentioned above and applied with the logic in my start comment works so far for me, however I'm absolutely not a .net or POS expert.

kodejack, 2020-01-28:
Updated PR #37 to handle disconnect, happy for comments on the approach, also added a flag to enable/disable reconnect on time out (Network only).

oachkatzalschwoaf, 2020-02-01:
I've only done a quick test a few days ago, and everything seems good so far.

oachkatzalschwoaf, 2020-02-10:
fixed in master

closed 2020-02-10 by oachkatzalschwoaf

lukevp, 2020-02-11:
@oachkatzalschwoaf @kodejack  thanks for coordinating this, I really appreciate it!  I don't have a network printer so I'm glad you all were able to get this fixed and merged in.