lukevp/ESC-POS-.NET #39: Code128 barcode not printing

neon-dev, 2020-01-29:
CODE128 barcodes don't print correctly. I believe it's due to missing start bytes in front of the actual data.
According to https://www.epson-biz.com/modules/ref_escpos/index.php?content_id=128#gs_lk, data doesn't start with the text bytes, but 123, [65-67]:
> d = 0 – 127
> (However d1 = 123, d2 = 65 – 67)

So length must be increased by two bytes and above bytes must be added transmitted in front of the text bytes.

neon-dev, 2020-01-29:
Oh, sorry. Just learned this has already been reported in #31 and fixed for v1.0. How to obtain this version via NuGet? I'm using 1.0.0-beta which is the latest one available and still having this bug.

closed 2020-01-29 by neon-dev

lukevp, 2020-02-01:
@neon-dev I plan on getting an RC done by mid February at the latest and I’ll ping you when it’s ready.

lukevp, 2020-02-01:
In the meantime you can just clone the repo from github and in VS add an existing project and point it at ESCPOS_NET which is the class library. Then you can switch branches to verify it’s fixed and once the RC is out you can switch back to nugget.