lukevp/ESC-POS-.NET #41: Does not print accents

ChipiCoding, 2020-02-14:
I have Encoding "IBM865"

Á Í Ó Ú upper case => does not print 

á é í ó ú =>  print ok

Encoding = Encoding.GetEncoding("IBM865")

byte[] IPrinterTemplate.Receipt(ICommandEmitter e) =>
            ByteSplicer.Combine(
            e.CenterAlign(),
            e.SetBarWidth(BarWidth.Default),
            Encoding.GetBytes(PrintHeader(PrintSaleModel.Header.Company))
}

lukevp, 2020-02-19:
Is this related to the character encoding on the printer itself?  What printer do you have? I believe you need to set it using a separate character page command, we don't have support for that but it should be pretty straightforward to add.

lukevp, 2020-03-05:
@ChipiCoding  any update on this?

closed 2020-04-19 by lukevp