lukevp/ESC-POS-.NET #44: Working with USB epson printer TM-T88V

juanMorenoDev, 2020-04-07:
Does this library suport USB connected printers? I'm trying to get the paper sensor status from a TM-T88V printer, can I do this with this library?

lukevp, 2020-04-07:
Good afternoon!

Yes, the library works great with USB-connected printers as long as you are using the virtual com port drivers provided by Epson.
I have updated the Readme with a USB Usage Guide, please review it and if you run into any issues, let me know on this issue and I'll update the guide.

lukevp, 2020-04-07:
Also, printer statuses are implemented in 2 ways, one is to read status as needed, but if you have an Epson, you can enable Automatic Status Back, and you can hook up an event every time a status changes (for example as paper starts and stops feeding, or runs out of paper, or door is open, etc. you will get a callback on your event with properties on what changed).  The included support app shows off the status changes and what data is in them.

lukevp, 2020-04-09:
@juanMorenoDev pinging to make sure you see this update.  I'm closing this issue but please let me know if you give it a try.

closed 2020-04-09 by lukevp

juanMorenoDev, 2020-04-09:
Yes! Thank you very much for your quick response, I test it and works fine!