lukevp/ESC-POS-.NET #46: Can't be installed

San-Nabaty, 2020-05-11:
Severity	Code	Description	Project	File	Line	Suppression State
Error		Unable to resolve dependency 'SixLabors.ImageSharp'. Source(s) used: 'nuget.org', 'Microsoft Visual Studio Offline Packages'.

lukevp, 2020-05-15:
Hey @San-Nabaty are there any versions of six labors that are available for you on nuget? What platform are you targeting?

I wonder if this is related to their recent licensing change...

San-Nabaty, 2020-05-17:
Im targeting '.NETFramework,Version=v4.7.2'

lukevp, 2020-05-21:
@San-Nabaty this looks like an issue with the dependency chain in SixLabors.ImageSharp in the version of the package we are using.  I was able to reproduce this issue, but upgrading the dependency to the RC version of 1.0 for ImageSharp resolved.  I have pushed an updated nuget package.  Can you please try again? Here is the new package (Currently validating but should be published in a few hours) https://www.nuget.org/packages/ESCPOS_NET/1.2.0

lukevp, 2020-05-28:
@San-Nabaty  pinging you for an update, I am closing this issue since this should be resolved but please let us know either way, and if it's not and we can reopen!

closed 2020-05-28 by lukevp