lukevp/ESC-POS-.NET #47: Windows printer share

mcarey1590, 2020-05-27:
Any thoughts on how to use this library to print via network share (\\%COMPUTERNAME%\Receipt Printer"), like the WindowsPrintConnector in this PHP library https://github.com/mike42/escpos-php

lukevp, 2020-05-27:
@mcarey1590 i haven’t tried that but based on a cursory glance of the php lib, it looks like it’s just writing to a file handle that’s a network share? You may be able to use the FilePrinter module in this library to do this. Let me know if you try this out!

mcarey1590, 2020-05-27:
Thanks for the suggestion. I'm going to try and get it working today. I'll let you know what ends up working.

mcarey1590, 2020-05-27:
With some modifications to the FilePrinter, I was able to get it mostly working. https://github.com/mcarey1590/ESC-POS-.NET/commit/11d542b9744d1fd9e36fff9be9fd6dc470477654

I haven't been able to test out this library with other printers, but with this method, using the ConsoleTest, It is always one command behind and the last command doesn't print until you close the connection and it clears the buffer. I'll mess around with it some more, but if that's the only issue I can work around that.

lukevp, 2020-05-28:
Hey @mcarey1590  that's awesome!  

Your error sounds like it's not flushing the buffer properly on write.  The BasePrinter type already debounces and properly flushes on appropriate intervals, but my guess is that calling flush on your binarywriter is not flushing properly to the underlying stream since you are using a direct file handle created using unmanaged code through p/invoke.  

I can think of 2 approaches: 1 - The Flush method is virtual, so you can override it and implement your own Flush method.  Just make sure you do the other parts:

```csharp
            BytesWrittenSinceLastFlush = 0;
            FlushTimer.Stop();
            Writer.Flush(); // this could be replaced with your own implementation
```

2 - you could do some more research on whether you can do this with managed code.  Then the flushing would probably work out of the box.  What did you run into that made you switch to the other approach?

Also, would you consider contributing this as a PR so we could have this functionality available for other users?  We would need to make sure that this will compile properly within a .net standard project and won't prevent the library from running on OSX and Linux, but this seems like a valuable use case that we should cover.

mcarey1590, 2020-05-29:
@lukevp I will hopefully have more time to work on this next week

I've already confirmed this code only works on windows because of the unmanaged code. This was able to solve an immediate need on a windows machine, but I will ultimately need to find a solution that supports OSX and Linux.

When i tried to use the FilePrinter, it said it could not find the file specified. When I changed the FileMode on the File.Open from `FileMode.Open` to `FileMode.Create` i got the following

```
Unhandled exception. System.NotSupportedException: FileStream was asked to open a device that was not a file. For support for devices like 'com1:' or 'lpt1:', call CreateFile, then use the FileStream constructors that take
an OS handle as an IntPtr.
   at System.IO.FileStream.Init(FileMode mode, FileShare share, String originalPath)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share)
   at System.IO.File.Open(String path, FileMode mode)
   at ESCPOS_NET.FilePrinter..ctor(String filePath) in f:\repos\ESC-POS-.NET\ESCPOS_NET\Printers\FilePrinter.cs:line 13
   at ESCPOS_NET.ConsoleTest.Program.Main(String[] args) in f:\repos\ESC-POS-.NET\ESCPOS_NET.ConsoleTest\Program.cs:line 89
```

lukevp, 2020-05-29:
I might have some time to help investigate this over the next couple of days.  What path are you using to open the file?  Could you provide me some directions on how to set up the printer this way in Windows?  I'd be happy to see if I can reproduce and maybe get it working under managed code.

mcarey1590, 2020-05-30:
I followed this article https://mike42.me/blog/2015-04-getting-a-usb-receipt-printer-working-on-windows

Basically setup your printer like normal on windows. Then go to properties/share and enable network sharing on the printer. The file path is then \\HOSTNAME\PRINTERNAME

igorocampos, 2020-06-03:
Hey people, may I give another suggestion about this?
Maybe it won't be the best approach, but I confirm this works (at least on windows):

```cs
byte[] data; //fill the bytes you want to send to the printer
string tempFile = "file.tmp";
string printerAddress = @"\\127.0.0.1\printer_shared_name";
File.WriteAllBytes(tempFile, data);
File.Copy(tempFile, printerAddress, true);
```

lukevp, 2020-06-03:
I actually tried this over the weekend actually, and I made a SambaPrinter printer type that did the file copy like you have.  The advantage is that it works without unmanaged code.  I couldn't get it to work without writing a temp file though, but it did work.  The problem was if you put it as an overload of the Write() method of the printer, it creates a BUNCH of files because the writes are meant to be buffered by the stream and flushed automatically, so they get called a bunch.  To make this performant so it's not writing a ton of temp files, it would need to implement its own throttled flush or something.

lukevp, 2020-06-03:
Here is the SambaPrinter implementation I was testing: 
```csharp
using System;
using System.IO;

namespace ESCPOS_NET
{
    public class SambaPrinter : BasePrinter
    {
        private string _filePath;
        private string _tempFileBasePath;
        private string _tempFilePath;

        public SambaPrinter(string tempFileBasePath, string filePath) : base()
        {
            _tempFileBasePath = tempFileBasePath;
            if (!Directory.Exists(_tempFileBasePath))
            {
                Directory.CreateDirectory(_tempFileBasePath);
            }
            _filePath = filePath;
        }
        public override void Write(byte[] bytes)
        {
            _tempFilePath = Path.Combine(_tempFileBasePath, $"{Guid.NewGuid()}.bin");
            File.WriteAllBytes(_tempFilePath, bytes);
            File.Copy(_tempFilePath, _filePath);
            File.Delete(_tempFilePath);
        }
    }
}
```

lukevp, 2020-06-03:
This version is way more performant than the previous, but it still flushes way too often for this type of file.  Maybe we could make the base printer have an optional parameter on the flush size and frequency that we could configure and crank up a lot higher for this printer, so for typical receipts it would just flush once?

```csharp
using System;
using System.IO;
using System.Timers;

namespace ESCPOS_NET
{
    public class SambaPrinter : BasePrinter
    {
        private string _filePath;
        private string _tempFileBasePath;
        private string _tempFilePath;
        private MemoryStream _stream;

        public SambaPrinter(string tempFileBasePath, string filePath) : base()
        {
            _tempFileBasePath = tempFileBasePath;
            if (!Directory.Exists(_tempFileBasePath))
            {
                Directory.CreateDirectory(_tempFileBasePath);
            }
            _stream = new MemoryStream();
            Writer = new BinaryWriter(new MemoryStream());
            _filePath = filePath;
        }
        public override void Flush(object sender, ElapsedEventArgs e)
        {
            FlushTimer.Stop();
            if (BytesWrittenSinceLastFlush > 0)
            {
                var bytes = _stream.ToArray();
                _stream = new MemoryStream();
                Writer = new BinaryWriter(_stream);

                _tempFilePath = Path.Combine(_tempFileBasePath, $"{Guid.NewGuid()}.bin");
                File.WriteAllBytes(_tempFilePath, bytes);
                File.Copy(_tempFilePath, _filePath);
                File.Delete(_tempFilePath);

            }
            BytesWrittenSinceLastFlush = 0;
        }
    }
}
```

lukevp, 2020-07-01:
@mcarey1590 are you still working on this?  I just pushed branch 47-add-support-for-network-printers if you'd like to give it a try.

mcarey1590, 2020-07-08:
@lukevp Thanks, I will try it out as soon as I get a chance. Been pretty crazy on my end.

lukevp, 2020-07-09:
No problem, I marked this as a requested enhancement so I will leave it open until we have a good solution for this use case.

lukevp, 2020-11-29:
hey @mcarey1590 wanted to check in on this issue to see if you've had a chance to test it. Also, I noticed in your profile that you're in Tyler, TX.  I'm in the Austin, TX area. Hope all's well with you, and feel free to add me on LinkedIn if you'd like to connect!

dredgy, 2022-01-01:
Did the samba printer ever get merged?

nickcharlton, 2022-01-26:
I just opened #159, which takes @lukevp's original branch and allows it to apply cleanly (since then the project was adjusted a little). My main need here has been to open up `Flush`, as I've been implementing something similar that prints directly using the Windows printer driver.

I'd love it if someone could test this out though; I'm not currently in a position to do this easily.

dredgy, 2022-01-26:
I will test today!

Joshua Dredge
0418251536 | ***@***.***

dredgy, 2022-01-27:
@nickcharlton I have tested this on a SEWOO SLK-TE212 and working perfectly :)

PR #159 (2022-07-08): Add support for Samba-based network printing
This adds support for printing via Samba-shared printers, along with an
additional option for the ConsoleTest example. Opening up
`BasePrinter`'s `Flush` method also allows for other extensions too.

This is @lukevp's prior work, fixed to allow it to apply cleanly.

https://github.com/lukevp/ESC-POS-.NET/commit/006bff2d63c21e13bc08b8ad9d6b88d008595ccc

Closes #47
commit 5089195d (2022-07-08): Add support for Samba-based network printing

This adds support for printing via Samba-shared printers, along with an
additional option for the ConsoleTest example. Opening up
`BasePrinter`'s `Flush` method also allows for other extensions too.

This is @lukevp's prior work, fixed to allow it to apply cleanly.

https://github.com/lukevp/ESC-POS-.NET/commit/006bff2d63c21e13bc08b8ad9d6b88d008595ccc

Closes #47
closed 2022-07-08 by lukevp