lukevp/ESC-POS-.NET #48: Code128 Type C incorrect encoding

dilucide, 2020-05-28:
Just for your information: The TM-T88x models of EPSON printers (and probably others, too) do not automatically encode CODE128 type C. With your library, Type C barcode would be printed as 48495051 for a barcode that should print as containing 1234.

The quick-and-dirty solution for this would probably look something like this (in ESCPOS_NET/Emitters/BarcodeCommands.cs):
```csharp
if (type == BarcodeType.CODE128) {
  if (code == BarcodeCode.CODE_C) {
    // Type C must have a length divisible by two.
    if (barcode.Length % 2 != 0)
      throw new ArgumentException(nameof(barcode));

    // Convert the input string to bytes and validate it to contain numerals only.
    // The input validation could be integrated into the BCD encoding loop; this was omitted for clarity.
    byte[] b = Encoding.ASCII.GetBytes(barcode);
    for (int i = 0; i < b.Length; i++)
      if (b[i] < '0' || b[i] > '9')
        throw new ArgumentException(nameof(barcode));

    // Convert the input string to BCD
    byte[] ob = new byte[b.Length / 2]; int obc = 0;
    for (int i = 0; i < b.Length; i += 2)
      ob[obc++] = (byte)(((b[i] - '0') * 10) + (b[i + 1] - '0'));

    // Replace the original input string with the BCD encoded variant
    barcode = Encoding.ASCII.GetString(ob);
  }

  barcode = barcode.Replace("{", "{{");
  barcode = $"{(char)0x7B}{(char)code}" + barcode;
}
```

lukevp, 2020-05-28:
Thanks for the info! Would you mind opening a PR with this change?  Would be happy to include it and release new version on nuget provided it works also with other printer types.

lukevp, 2020-05-28:
Do the image tests and all other barcodes work properly on your printer other than this type?

dilucide, 2020-05-29:
> Do the image tests and all other barcodes work properly on your printer other than this type?

Hi,

I'll do a PR during the weekend.

I haven't actually tested your library yet, but stumbled on your library trying to resolve issues printing long type C Code128 barcodes with my library. My test case was a twenty digit number which would be far to wide when not encoded properly, thus my barcodes would not print. Only after having seen the implementation of your library sending the exact same data to the printer, I suspected a length issue and found the BCD encoding problem. I'll install your library during the weekend and report back.

lukevp, 2020-05-29:
That sounds great!  Please give it a try, you can just clone the repo and run ConsoleTest, then try doing the image and the barcode printing tests.  

The reason I asked is because it's possible to have the driver configured in such a way where certain escape sequences don't work properly and others do, and I want to make sure it's really just this one affected (which would point to a data incompatibility) vs a configuration issue with the driver itself.  What OS and driver setup are you using for your printer?

dilucide, 2020-05-29:
I've just tried your library after having implemented the bug-fix. The test prints work flawlessly. I took the liberty to correct a minor issue with your console test program not allowing to change the serial port and added a test case for CODE128 type C.

I am using a EPSON TM-T88V running under Linux and Windows 10 each with their corresponding EPSON drivers. I do believe this "issue" is by design as I observe the same behavior on an old TM-T88III on a physical RS232 port.

lukevp, 2020-05-29:
Sounds great, looking forward to the PR! I’ll test it on the newer Epson I have to make sure there are no regressions. Glad you were able to print from multiple OSes too!

dilucide, 2020-06-01:
Is the PR satisfactory?

dilucide, 2020-06-26:
I've corrected the issue. See the PR for details.

PR #49 (2020-07-01): Added quick-and-dirty bug-fix for CODE128 Type C.
Fixed small issue in test application not allowing to change the serial port.
Fixes #48
closed 2020-07-01 by lukevp