lukevp/ESC-POS-.NET #50: Nuget Download Error

airmikec, 2020-06-01:
I tried to download your code from Nuget and I got this error

Unable to resolve dependency 'SixLabors.ImageSharp'. Source(s) used: 'nuget.org', 'Microsoft Visual Studio Offline Packages'.

lukevp, 2020-06-01:
Hey @airmikec, I got a report for a similar issue on #46 , could you read it over and come back and let me know what's going on for yours?  What Nuget version of ESCPOS_NET are you trying to install, and what version of .NET/.NET Core are you using?

airmikec, 2020-06-01:
Thank you for the quick response.  Here is more details:

NUGET Version : 1.2.0
Initial .NET Version : 4.6
I changed the project to .NET Version : 4.7 and 4.8

I still get the error. I also read issue #46 and it didn't really help much. All that I see is someone had the same error you fixed it and that is what published for version 1.2.0. Is it possible that there is not proper binding to `SixLabors.ImageSharp`?

lukevp, 2020-06-01:
@airmikec Looks like there were 2 issues with older .net apps, the first one was fixed with 1.2 but the other is because SixLabors.ImageSharp is still a prerelease package.  

I apologize for the issues, you are not supposed to take dependencies on prerelease packages, but unfortunately this is the library that this lib currently depends on for image manipulation.  In the future, we could port it to System.Drawing, but ESCPOS_NET was started when that version did not exist for .net core.

I reached out to SixLabors several months ago asking them to promote their package to a production version - it is already been downloaded 5 million times!  They still haven't done so, so unfortunately you have to enable prerelease packages to be able to install it. Check the Include prerelease and it should work.

[image]

airmikec, 2020-06-01:
Sorry. Checking pre-release did not work. I attached an image to this so that you can see what I have done.

[image]

airmikec, 2020-06-01:
So it occurred to me that you said something about pre-release and the software was in pre-release. So I actually downloaded SixLabors.ImageSharp and then I installed your product and that did work.  So now I got it working.

lukevp, 2020-06-02:
@airmikec I tried this earlier today and it does seem like the dependencies are not resolving properly.  You can still use nuget and just install SixLabors.ImageSharp directly and then install ESCPOS_NET and it will work fine, but I'm not sure why it's not resolving the packages correctly.

airmikec, 2020-06-03:
Did that.. we are good to go.. Thank you.. :D

closed 2020-06-03 by airmikec