lukevp/ESC-POS-.NET #51: Any chances of supporting Bluetooth classic/LE on Android/iOS platforms?

adli89, 2020-06-03:
This would be the perfect package that can integrate with everything if Bluetooth classic/LE is supported. Is there a plan for that?

igorocampos, 2020-06-03:
You can use this library to get the corresponding `byte[]` you want to print and then connect yourself via Bluetooth and send the data. Let me show how I did so in one of my Android projects:

```cs
private static string bluetoothPrinterAddress = "00:01:02:03:0A:0B"; //replace with your BT Address
private static UUID PRINTER_UUID = UUID.FromString("00001101-0000-1000-8000-00805F9B34FB");

private static BluetoothDevice getDevice(BluetoothAdapter bluetoothAdapter)
{
    foreach (BluetoothDevice device in bluetoothAdapter.BondedDevices)
        if (device.Address == bluetoothPrinterAddress)
            return device;
    
    return null;
}

private static BluetoothSocket getSocket(BluetoothDevice device)
{
    BluetoothSocket socket = device.CreateRfcommSocketToServiceRecord(PRINTER_UUID);
    socket.Connect();
    return socket;
}

private static void sendData(byte[] bytes, BluetoothSocket socket)
{
    var o = socket.OutputStream;
    o.Write(bytes, 0, bytes.Length);
    o.Close();
}

public static void Print(byte[] byteArray)
{
    BluetoothAdapter btAdapter = getBTAdapter();
    
    //Bluetooth is turned off
    if (btAdapter is null)
        return;

    BluetoothDevice device = getDevice(btAdapter);

    //Printer not found
    if (device is null)
        return;

    BluetoothSocket socket = getSocket(device);
    sendData(byteArray, socket);
}

```

lukevp, 2020-06-03:
@igorocampos  this is awesome, I would love to support this since the library supports basically all other printing types already!  Do you think you could implement this as a BluetoothPrinter type like we have for the other printer types?  How are you using Bluetooth from within the Android namespace, is it built against xamarin or something?  I wonder if we could do this in a cross-platform way?

igorocampos, 2020-06-03:
@lukevp this code example was from a xamarin project alright. I don't own the devices anymore, but they were an Android POS with a BT printer embedded. But once you get the Bluetooth socket from whatever hardware you have, it should be straight forward to send the bytes over.

adli89, 2020-06-04:
Yes, this is what I was planning to implement. Just asking if there are plans to support it right in the library. Thanks for the code snippet, will definitely use this library for my purpose. Great library 😁

closed 2020-06-04 by adli89

igorocampos, 2024-01-23:
> @igorocampos this is awesome, I would love to support this since the library supports basically all other printing types already! Do you think you could implement this as a BluetoothPrinter type [...]

Coming back at this one again, @lukevp would you consider haveing [32feet](https://github.com/inthehand/32feet/tree/main)as a dependency for BT comms?

If yes, I think we can definitely do a cross-platform Bluetooth printer, what do you think?