lukevp/ESC-POS-.NET #52: QRCode support

igorocampos, 2020-06-03:
I took a look at your classes and couldn't find a command to print QRCodes, is this not implemented yet, or did I miss it?

If not, want me to do a PR about it? Just tell me where would be a good place to do so in the code :)

lukevp, 2020-06-03:
A PR would be great! I've received a couple PRs for QR code support, it is not a whole lot of work but we haven't gotten one merged yet because the library had a lot of refactoring before V 1.0.  Take a look at these that were done on an older version.

I think these should be implemented similarly to Barcodes, but with a name like 2DCodeCommands
and similar enums and validators as the barcodes have.  If you look at DataValidation/DataConstraint and DataValidator they provide some useful guardrails for users to make sure they don't use the call incorrectly.  The reason I did this is that you can send the printer bad data and it will spit out random characters so I thought this was a nice thing to have in the library.

https://github.com/lukevp/ESC-POS-.NET/pull/11
https://github.com/lukevp/ESC-POS-.NET/pull/19

igorocampos, 2020-06-03:
Cool, I'll take a look and see if I can come up with a good PR.
Yeah, I know what you mean about printing bad data because of invalid byte parameters

ssoellinger, 2020-07-03:
@lukevp : What was the problem with #11 ?, I tested the Code and it work well

igorocampos, 2020-07-03:
@ssoellinger Oh man, I just pushed #61!! The timing... But feel free to go there and propose any changes :)

@lukevp Sorry it took me so long to code this, I had a way too busy month (contrary on what I thought I would have when I proposed to do the PR).

lukevp, 2020-07-04:
no problem @igorocampos and thanks for contributing! It's very appreciated.  I'll comment on  #61.

lukevp, 2020-07-07:
Issue closed, PR #61 has been merged. Thanks @igorocampos !

closed 2020-07-07 by lukevp