lukevp/ESC-POS-.NET #53: Suggestion about ByteSplicer

igorocampos, 2020-06-03:
Hi Luke, 
May I suggest the ByteSplicer Combine method accept only `byte[]` instead of `object[]` as a parameter?

https://github.com/lukevp/ESC-POS-.NET/blob/554802bb73e2de5b229c441468b013ae5fefa3d9/ESCPOS_NET/ByteSplicer.cs#L7-L20

I mention it, because it looks like this method is not ready to use other type of params. That would simplify a little bit your code and avoid potential exception to be thrown

```cs
public static byte[] Combine(params byte[] byteArrays)
{
    ByteArrayBuilder builder = new ByteArrayBuilder();
    foreach (var byteArray in byteArrays)
        builder.Append(byteArray);
    return builder.ToArray();
}
```

lukevp, 2020-06-03:
Hey @igorocampos thanks for the feedback.  The reason why it is a params object[] is because if it is a params byte[] byteArrays, you can only pass bytes (eg. 
```

lukevp, 2020-06-03:
As soon as a posted this reply I realized the issue, it's because params defines an array type itself, so the proper implementation is `byte[][]`
```csharp
public static byte[] Combine(params byte[][] byteArrays)
        {
            ByteArrayBuilder builder = new ByteArrayBuilder();
            foreach (var byteArray in byteArrays)
            {
                builder.Append(byteArray);
            }
            return builder.ToArray();
        }

lukevp, 2020-06-03:
I'll update this in the next version, thanks for the feedback!

igorocampos, 2020-06-03:
Glad to help :)

Not sure if you would like to support a mix of byte[] and strings too. I've done that before because a user could append strings without having to worry about converting to byte[] before, for example something like:
```cs
ByteSplicer.Combine(e.CenterAlign(), "A CENTERED TEXT", e.LeftAlign(), "LEFT ALIGNED TEXT");
```
However not sure how interesting this would be for you.

lukevp, 2020-06-03:
I'd like to do that but the Print method of the emitter is responsible for sanitizing strings and fixing newlines in the strings, so you could do 
```csharp
ByteSplicer.Combine(e.CenterAlign(), e.Print("A CENTERED TEXT"), e.LeftAlign(), e.Print("LEFT ALIGNED TEXT"));
```
but the byte splicer does not itself know what emitter you're using, so we can't assume to know how to convert the text to proper print commands in the future if they are not able to be emitted directly as text (say for a different type of thermal printer).  

Perhaps this could be a good use case for a fluent API like 
```csharp
var DirectPrinter = new DirectPrinter(emitter, printer);
DirectPrinter.CenterAlign()
                    .Print("Centered Text")
                    .LeftAlign()
                     .Print("Left Aligned Text")
                    .Send();
```

and the .Send() method would internally maintain a byte array and call bytesplicer.combine and send on to the underlying printer whenever it's desired?  This would work more like some other people's libraries do but still have the composability of different emitter and printer types.  The downside is the DirectPrinter would have to wrap every single command with the same signatures as the underlying emitter.

igorocampos, 2020-06-03:
Hmm, I see... There is sanitation before converting the string to byte[]. And this is going to be different depending on the emitter, is that it?

Yeah, in that case I agree with you

PR #80 (2020-10-01): better performance of ByteSplicer
Fixes #53
closed 2020-10-01 by lukevp