lukevp/ESC-POS-.NET #56: printing from blazor webassembly

fabianus76, 2020-06-09:
Hi Luke, 

excuse me for perhaps a nonsense question: Is it possible to implement your library in a blazor webassembly application and print to a network printer ?

Regards, 
Fabianus

lukevp, 2020-06-09:
Hey Fabianus,

The byte builder part of the library should all be usable. The network printer may not work since it is expecting to be able to open direct TCP connections. The image processing functions in image sharp probably won’t work but maybe they will?

Typically if you are making a web app and you want local printing support, you would implement this by building a local webserver and then making RPC calls to it from your web app to local host, and that server would have privileges to connect to serial ports and such. This is how other solutions accomplish this, eg:

https://www.neodynamic.com/articles/How-to-print-raw-ESC-POS-commands-from-Javascript

https://www.printnode.com/en

This should be pretty easy to do with .net core and kestrel, but would require the end user to run an app on their device like these other solutions. If you want to try to accomplish this without that setup you will need to be able to make http or https calls to the printer, which is unlikely if you’re trying to print esc/pos.

If this is a use case you really want to explore, perhaps as a commercial endeavor, I’d love to help out and would be available to build this out if you’d like. We could also do this open source too. It would be a helpful extension to this lib.

fabianus76, 2020-06-11:
Hey Luke thanks a lot for all the details!
I think this is a really fundamental question as printing from web applications to pos printers is probably one of the most common use cases of printing from those applications. Having to install a local printing server is a detour that creates all kinds of troubles. If a blazor webassembly application would be capable to print tickets, bills etc. without any additional preparation - wow. 

Your write that making http calls to the printer is unlikely f or print esc/pos. I am not sure to understand all implications but wouldn't that be possible without any detour with this model ?

https://www.epson.eu/viewcon/corporatesite/products/mainunits/overview/11357#specifications

lukevp, 2020-06-13:
Fabianus,

I researched this more, and I was wrong about this.  If you wanted cross-platform support for TCP/IP printing (against a server) that is already built into the library.  However if you want to print directly from a browser to a receipt printer without any software in between, this is in fact possible with Epson printers.  The TM intelligent printers have a built in server that you can directly connect over http(s) using the Epson oPOS Javascript SDK.  The printer you linked would NOT support this, as it is a normal dumb network printer, which will require native functionality.  Here is a list of "intelligent" printers that have the embedded webserver: https://download.epson-biz.com/modules/pos/index.php?page=genre&pcat=52

If you wanted to pursue this, we would love to have a PR!  I don't have a TM-Intelligent printer so I can't assist with development, but it seems possible to create an interop between Blazor and the Epson JS API and then print directly to the printer that way with nothing else in between.  

This would be epson-specific and would only support these TM-I printers.  If you did it with the separate webserver as I mentioned before, it would be universal for all ESC/POS printers (even cheap non-epson printers).

lukevp, 2020-06-13:
@fabianus76  one other thing to keep in mind would be the fact that a $35 Raspberry Pi can be used for this webserver - the library works great on Raspberry Pis directly to USB printers - and you would be able to buy MUCH cheaper printers then as you would be able to use USB printers.  Something to keep in mind, as the price of these intelligent printers (and network printers in general) is generally much more than $35.  This would also give you compute on the edge which would let you potentially do other POS-related operations.  Not sure what you're building or your business model but something to keep in mind.

lukevp, 2020-06-13:
Here is something you could use to run on the raspberry pi if you wanted to pass through the bytes directly from a websocket to a TCP connection, but will still require something like a raspberry pi to run it: https://github.com/novnc/websockify

fabianus76, 2020-06-17:
Hey Luke, what should I say. I am amazed about the research and very thankful for all the information you put together. 
Both ideas (TM intelligent printers and an intermediate raspberry) are very interesting. I agree that Epson printers are not the cheapest, but from my long term experience they are worth the money (I have 4 TM-T88V in production since 10 year - not one incidence so far). 
I would love to tell you more about my project and will PM you.

closed 2020-06-17 by fabianus76

lukevp, 2020-06-19:
@fabianus76  happy to help!  Where do PMs go in GitHub?  I'm not sure if you haven't messaged me yet or if I just don't know where they go.

fabianus76, 2020-06-20:
Thanks Luke, I tried to find your e-mail address on https://www.paireepinart.com/ but couldn't find it there. Would you mind to send me yours at fabianus (a) gmail ?