lukevp/ESC-POS-.NET #59: Code-128 CODE-C Barcode Support

lukevp, 2020-07-01:
@dilucide  based on the review of this doc: https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=128
It looks like CODE-C should support alpha characters in addition to numeric, is there a reason that your PR had a validator that only allowed numerics?

dilucide, 2020-07-02:
If you look at ISO 15417 (or the table in the aforementioned reference under heading **[Notes for CODE128 (m = 73) process]**) you'll see that Code Set C encodes the digits as BCD, thus per byte two numerals are encoded. In contrast to Code Sets A and B, Code Set C does not support alphanumeric characters. This is why you cannot have a content length that is not divisible by two.

For clarification: If you wanted to encode the numerals 12345678, these would be encoded as 0x3132333435363738 using Code Set A or B in contrast to Code Set C in which 0x12345678 would be encoded.

-----

Regarding your comment on the PR: Do you still want me to update to explicit block syntax and add a comment? I've downloaded the files I found to be problematic directly from GitHub and found them to have consistent line endings. Where did you notice additional issues?

lukevp, 2020-07-04:
No worries, I just pushed a new version with the more explicit block syntax.  As for the newlines, I think this was my error and that these were introduced by previous PRs, and your PR fixed them.  I just saw the line ending changes in the PR and thought that it was yours but I think yours was correcting other PRs.

I've done a few refactors on the code, could you pull master and make sure I didn't break it?  If it's still working I'll push this new release to Nuget.

dilucide, 2020-07-06:
The master looks fine. 

I've just noticed an error that is persistent pre-refactor and post-refactor, though: The FilePrinter does not support creating a file, but you take this as a prerequisite in the large byte array test. A simple fix would be to change ```FileMode.Open``` to ```FileMode.OpenOrCreate``` in Line 14 in FilePrinter.cs of ESCPOS_NET.ConsoleTest.

dilucide, 2020-07-06:
And there is an oversight on my part in TestBarcodeTypes.cs in Line 23: As a result of what can only be a typo, the test string for CODE128 Code Set C is not of even length.

lukevp, 2020-08-02:
@dilucide  The FilePrinter is used on unix-like OSes because the serial port is exposed as a file:
```
                else if (RuntimeInformation.IsOSPlatform(OSPlatform.Linux))
                {
                    Console.Write("File / virtual com path (eg. /dev/usb/lp0): ");
                    comPort = Console.ReadLine();
                    if (string.IsNullOrWhiteSpace(comPort))
                    {
                        comPort = "/dev/usb/lp0";
                    }
                    printer = new FilePrinter(filePath: comPort);
                }
```
In that case we probably don't want to create the file if it doesn't exist.  I went ahead and added a default-false boolean called createIfNotExists that will call OpenOrCreate to support that, and used it in the FilePrinter test.  Let me know if you have any other thoughts on that implementation, I'm going to go ahead and close this issue.

The TestBarcodeTypes typo was resolved by a separate PR @paul-turner-github  opened #65 that has been merged.

closed 2020-08-02 by lukevp