lukevp/ESC-POS-.NET #60: Support for Scanning check

cashnetpartha, 2020-07-02:
Hi, I am using EPSON TM-T88V and i want to create a sample .NET core console application for scanning the check. are you supporting this.

lukevp, 2020-07-02:
Hey @cashnetpartha  thanks for checking out the library! 

As far as I can find, the standalone TM-T88V does not support check scanning or validation. 

Could you link me to the exact model on the Epson site that you are using? Perhaps it's [something like this](https://epson.com/For-Work/Point-of-Sale/POS-Printers/TM-S9000-Multifunction-Teller-Device/p/A41A267011) that has integrated TM-T88V printer included with the scanner?

Are you trying to scan the MICR data off the checks, scan the actual check in as an image, or just print validations on them?  

There are not very many receipt printers that support check scanning, but there are quite a few that can print validations on the back of the check.  

I would be interested in helping to support whatever hardware you have, but I don't have any check scanners myself.  Would you be able to provide some hardware to me for testing purposes?

cashnetpartha, 2020-07-03:
Thanks for answering. i am working on TM-T88V and as you mentioned it is only for printing receipt. Please ignore my comment on check scanning which is not required as of now.

closed 2020-07-03 by cashnetpartha