lukevp/ESC-POS-.NET #63: Invalid argument when creating SerialPrinter object on Linux (Ubuntu 14.04)

KyryloAntoshyn, 2020-07-08:
Hi, I am trying to leverage you library in Ubuntu 14.04.

I have printer connected via USB. To communicate with it I try to use SerialPrinter class. According to the documentation I pass "/dev/ttyUSB0" file and 9600 baud rate. This file and baud rate are for my printer, I rechecked it. However, when I try to create SerialPrinter object I have exception: Unhandled exception. System.IO.IOException: Invalid argument. I use code from the documentation steps.

[image]

Why does it happen and how can I fix it?

Thanks in advance!

lukevp, 2020-07-08:
That’s great that you’re trying it out Ubuntu! I haven’t tested the library
on Linux in quite a while, since it used serialportsrream instead of
system.io.ports, but we should be able to get it working.

Can you provide the code you are using to test? Have you tried pulling the
code from GitHub and using the console test app?

Are you testing from an x86 machine or an ARM like a raspberry pi?

It sounds like your params to the serial port constructor may be messed up.

Any reason you’re using 9600? For USB you can use a much higher baud rate
typically (I use 115200). It should speed up your printing quite a bit.

KyryloAntoshyn, 2020-07-08:
I fixed the issue today and everything works. I changed my 25-pin–usb cable to usb-usb and used FilePrinter class like you did in ConsoleTest application. It works fine now. ConsoleTest also works fine.

I had my error on ARM Ubuntu 14.04. I used plain code from the readme of the library:
[image]
I choose 9600 baud rate, because system provided me this value for /dev/ttyUSB0 port.

KyryloAntoshyn, 2020-07-08:
I closed the issue, thanks for your attention. If you have more questions about the issue ask me here and I will try to answer.

lukevp, 2020-07-08:
Glad you figured it out!  That does sound familiar.  In case this affects you, the FilePrinter is write only, so you will not be able to get status messages back (eg. paper out, door open).  That is only implemented in the SerialPrinter right now.  If you need that functionality, we probably just need to change the serial printer setup to work on Linux too.  If so, reopen this and we'll look into it.

KyryloAntoshyn, 2020-07-09:
Yes, it is necessary for me to have the "empty paper" message in a production environment. I reopen this issue and we need to make SerialPrinter working on Linux. Thanks!

lukevp, 2020-07-09:
@KyryloAntoshyn could you confirm what version of the library you are using and if it is from Nuget or GitHub?  

Is it possible for you to try to build the Console Test project in this repo from the latest master/ and see if it works or not?

I won't have time to look at this myself until the weekend, but I can try and guide you through this.

KyryloAntoshyn, 2020-07-10:
I can do it next week and send you the app.

KyryloAntoshyn, 2020-07-15:
Hi, here is test app which references your library from GitHub. Exception occurs when I create SerialPrinter object. My printer's virtual port is /dev/ttyUSB0, you can see it in code.

[SerialPrinterTest.zip](https://github.com/lukevp/ESC-POS-.NET/files/4926073/SerialPrinterTest.zip)

lukevp, 2020-11-29:
hey @KyryloAntoshyn the lib has been updated a few times since this issue was open.  Could you check and see if this is still an issue with the latest version?

KyryloAntoshyn, 2020-12-02:
Hi, I've tested the library on Linux. FilePrinter works well with cutting!

KyryloAntoshyn, 2020-12-02:
Hi, @lukevp. I will test library as soon as possible.

KyryloAntoshyn, 2020-12-17:
I have downloaded latest NuGet library version. I have Epson TM-T70 printer connected via USB. When I am trying to create `SerialPrinter` object the error occurs:

Code:
`var printer = new SerialPrinter(portName: "/dev/usb/lp0", baudRate: 115200);`

Error:
[image]

KyryloAntoshyn, 2020-12-17:
Is it possible to use `FilePrinter` and check for paper status now? `FilePrinter` works well.

Rombersoft, 2021-09-12:
It is not problem. I always use FileStream class for USB ports printers. It does not need baudrate, parity, handshake and other shit.