lukevp/ESC-POS-.NET #64: Question network and USB print

arielbvargas, 2020-07-09:
Hi! Thanks for this excellent library. I was trying to implement it and make it work with an OCOM Thermal Printer. The network approach is not working since I don´t know the port (tried port 80 and 8080 with no sucess).
USB driver installs the printer on USB001 and when I try to configure it as defined in the documentacion I get an exception:
_The given port name does not start with COM/com or does not resolve to a valid serial port.
Parameter name: portName_

Can you please point me in the right direction to make it work through USB?
Thanks in advance!

lukevp, 2020-07-09:
Happy to help! What OS are you on?

arielbvargas, 2020-07-09:
> Happy to help! What OS are you on?
> […](#)

Hi! Windows 10. Thanks!

lukevp, 2020-07-09:
Have you seen this section of the readme that details installing the epson  drivers? USB on windows requires a virtual com port driver.

If you want to use wireless, the network example in the readme shows the standard port, it’s 9000 or 9100 or something.

https://github.com/lukevp/ESC-POS-.NET#usb-usage-guide

arielbvargas, 2020-07-09:
Hi @lukevp,
Yes, I´ve seen the section regarding de virtual com port driver. Unfortunately, this printer is an unknown brand (OCOM) and I can not find a way to install the virtual com port driver.
Regarding the EPSON printer, using the standard port I was able to connect through ethernet. 

Thanks for your help!

lukevp, 2020-07-09:
Yeah the virtual com port linked is Epson-specific, but you must have some type of COM emulation for the library to be able to connect.  

Unfortunately the solution will be device-dependent since this app does not work natively with USB.  Do you have any other connection type with the device?  Could you link me to the exact model on the OCOM page?

You can alternately give this other library a try.  It uses the windows print spooler to print, so it may work for your situation without the virtual COM port.  I don't personally like the way the library's structured with all methods off the printer object, but if it's your only option it may be worth a try.

https://github.com/mtmsuhail/ESC-POS-USB-NET

arielbvargas, 2020-07-10:
Hi @lukevp!

Yes, sure. The OCOM URL for this printer is: https://www.ocominc.com/products/OCPP-80G-URL-Reliable-80mm-Thermal-Receipt-Printer-With-Three-Interface.html

Thanks for the suggestion of using ESC-POS-USB-NET library. Fortunately, I was able to connect with the printer using an ethernet connection. Everything now works as expected.

lukevp, 2020-07-11:
Thanks @arielbvargas, I've closed the issue.  If you choose to try and add support for this via USB, the main thing you need to do is figure out how to get windows 10 to have a COM mapping of the serial data onto the USB.  If the printer manufacturer doesn't supply drivers for this, you can do this with commercial Virtual COM Port libraries.

closed 2020-07-11 by lukevp