lukevp/ESC-POS-.NET #66: How to print logo from a Xamarin Forms Application

N4511, 2020-07-12:
Hi, I am trying to print logo using this nuget package. Could not figure out how to print the logo yet.  I have tried getting  image byte code using the webclient.downloadData("ImageURL"). But it only prints gibberish instead of the logo itself. Could you be kind enough to share how to print a logo from Xamarin forms application. Thanks.

lukevp, 2020-07-14:
Hey @N4511 thanks for trying the library out.  Do you have an Epson printer?  Can you try printing from outside of the Xamarin app (directly on Windows) using the test app to see if it works correctly?

N4511, 2020-07-17:
Thanks for coming back to me so quickly. My printer is not a Epson printer. I should try to print using the test print. Thanks. But if you could figure out how to print from a xamarin forms would be highly appreciated.

N4511, 2020-07-23:
Just used the test app. And the result is still the same. I guess it's an issue on my printer side. Thanks.

lukevp, 2020-07-23:
@N4511 What printer do you have? What OS and Driver?

N4511, 2020-08-06:
The model of my receipt printer is BTP-R180ii and the make is SNBC. It uses its generic windows driver that has been provided by the maker. Thanks.

lukevp, 2020-08-07:
Could you put up an example Xamarin forms app? Or make a PR to this repo with another example? I can give it a try with my printer to see if I can find the issue.

lukevp, 2023-02-02:
closing due to staleness.

closed 2023-02-02 by lukevp