lukevp/ESC-POS-.NET #67: Problem printing Japanese text

riaanvdl, 2020-07-23:
Hi, this is a great library thanks, but I am having issues printing Japanese characters. My POS printer does support Katakana on code page 1.

This is what I have tried so far but it all ends up wrong. See attachment for the result. Any help with this would be greatly appreciated. Thanks a lot!

`var printer = new NetworkPrinter(ipAddress: Settings.PrinterIP, port: Settings.PrinterPort, reconnectOnTimeout: true);

            Encoding encodingKatakana = Encoding.GetEncoding(932); //Japanese encoding (I also tried Shift-JIS)
            Encoding unicode = Encoding.Unicode;

            byte[] utfBytes = Encoding.Unicode.GetBytes("こうという 気持 きも");
            byte[] convertedBytes = Encoding.Convert(unicode, encodingKatakana, utfBytes);

            printer.Write(utfBytes); //Attempt straight from Unicode

            var e = new EPSON();
            printer.Write(
              ByteSplicer.Combine(
                e.CodePage(CodePage.JAPAN) 
              )
            );

            printer.Write(convertedBytes);  //Attempt from Japan codepage

            printer.Write(
                ByteSplicer.Combine(
                e.CodePage(CodePage.KATAKANA) 
                )
            );
            printer.Write(convertedBytes); //Attempt from Katakana codepage`

[image]

lukevp, 2020-07-23:
Hey @riaanvdl  glad you like the library!  You are the first to attempt to print any CJK language as far as I know.  I would love it if we could figure out what the issue is and then contribute back to the library (either in the README or in the actual code?) but it may take some troubleshooting.

Are you able to print everything from the test app? 
What OS are you using?
Are you sure that the string used here is Unicode?  Unicode means UTF-16.  If it is a UTF-8 string, you need to use Encoding.UTF8 instead.

        `byte[] utfBytes = Encoding.Unicode.GetBytes("こうという 気持 きも");`
Could you read your bytes from a file using File.ReadAllBytes() from a file that has a UTF-8 encoding to make sure the encoding is consistent on our input data?

I am not very familiar with Japanese but I know that there is a significant difference in complexity between Kanji and Katakana.  That's why Katakana is the only printing option, because the printers have a very small space for font characters. In these receipt printers, the fonts are local to the device so they are not able to load an entire Kanji set.

Is this input string Kanji or Katakana?  If it's Kanji, we will probably have to convert it to Katakana before printing (encoding is basically just translating lookup #s, so it will not know that this Kanji character is these Katakana characters and map them).

riaanvdl, 2020-07-24:
Hey Luke, thanks for the reply. 
Your reply actually pointed me into the right direction (I think) and looks like I might get this working...
I will post the full solution as soon as I have it 100% working.

riaanvdl, 2020-07-26:
Just coming back here to post the solution for someone else that might need this in the future.
There were are couple of things wrong to start with.

The text I tried to print was translated to Japanese using Google translate.
This caused the first problem because it turns out that Google Translate mainly translates the English to Kanji. The POS printer that I have can only print Katakana. 
1) So I had to run the English to Japanese translation through Google translate, but then I had to convert the Kanji to Katakana using this translator: [https://nihongodera.com/tools/kana-converter](https://nihongodera.com/tools/kana-converter)
(I have no idea what the quality of the end translations are but I suspect a lot of the context was lost during the Kanji to Katakana translation, but there is not much I can do about that really)
2) The second problem was that the POS printer can only print half width Katakana chars, so printing the string from step 1 directly also printed "?". I found an awesome Nuget package that can convert the full width Katakana to half width Katakana.
[https://github.com/usausa/Smart-Net-Text-Japanese](https://github.com/usausa/Smart-Net-Text-Japanese)

Here is a code example:

`            
var printer = new NetworkPrinter(ipAddress: Settings.PrinterIP, port: Settings.PrinterPort, reconnectOnTimeout: true);

            Encoding enc = Encoding.GetEncoding("Shift-JIS"); //Japanese encoding
            Encoding utf8 = Encoding.UTF8;

            string Katakana = "ニモツ、 バッグ、 ニモツ、 ケース、 リュックサック、 ウバグルマ、 コート ハ、 フクスウ ノ バアイ モ アル シュウノウ アイテム ヲ サシマス。 タ ノ アイテム ハ ウケイレラレマセン。 イキテイル カ シンデイル ドウショクブツ、 レイゾウ ガ ヒツヨウナ ショクヒン、 キケン、 キケン、 コワレ ヤスイ、 ホウリツ デ キンシサレテイル モノ、 マタハ モレル カノウセイ ノ アル エキタイ ハ、 ホカン ノ タメニ ウケイレル コト ガ デキマセン。 バッグ ハ ロック、 セキュリティ タグ、 マタハ シール デ ホゴスル ヒツヨウ ガ アリマス。 バッグ ニ キチョウヒン ヲ イレナイデ クダサイ。 ミブン ショウメイショ、 リョコウ ショルイ、 リョコウ ニ ヒツヨウナ チケット ナド ヲ ニモツ ノ ナカニ イレナイデ クダサイ。 バッグ カバー ハ キホン カカク ニ フクマレテオリマセン。 バッグ カバー ハ サンカ サイト デ ツイカ リョウキン デ コウニュウ デキマス。 サンカ ノ ショウサイ ニツイテ ハ、 サイト ニ トイアワセテ クダサイ。 バッグ カバー ハ、 バッグ ヲ フンシツシタ バアイ ニ ノミ カバーシ、 バッグナイ ノ ココ ノ アイテム ハ カバーシマセン。 バッグ カバー ハ、 スベテ ノ アイテム ヲ コベツ ニ デ ハ ナク、 ゼンタイ ノ カバー トシテ カバースル 1 ツ ノ リョウキン デス。 トウチャクジ ノ バッグ ノ ソンショウ ノ ケンサ ハ オコナッテイナイ タメ、 バッグ ノ ソンショウ ニタイスル クレーム ハ オコナワレマセン。 バッグ ハ テキセツ ニ トリアツカワレ、 アンゼン ニ ホカンサレマス。 バッグ ノ ガゾウ ハ、 バッグ ヲ シキベツスル タメニ サツエイサレル バアイ ガ アリマス ガ、 バッグ ノ ジョウタイ ヲ キロクスル タメニ ハ シヨウサレマセン。 セキュリティ タグ ガ テイキョウサレテイル バアイ ハ、 セキュリティ タグ バンゴウ ノ シャシン ヲ トル コト ヲ オ ススメ シマス。 イチブ ノ バショ デ ハ、 セキュリティジョウ ノ リユウ カラ、 ウケイレ マエ ニ ニモツ ヲ アケテ ナカミ ヲ ケンサスル ヒツヨウ ガ アル バアイ ガ アリマス。 シュウシュウ ノ キゲン ガ キレル マエ ニ、 カナラズ バッグ ヲ シュウシュウシテ クダサイ。 ユウコウ キゲン ガ キレタ ノチニ バッグ ヲ カイシュウシタ バアイ、 サイト ハ ドクジ ノ サイリョウ ニヨリ、 カク アイテム オヨビ カク チエン キカン ニタイシテ ゼン キカン ノ リョウキン ヲ セイキュウスル コト ガ デキマス。 レシート ハ POS インサツ、 デンシ メール、 マタハ シャシン ノ レシート デス。 アイテム ヲ シュウシュウスル トキ ハ、 レシート ヲ ホカンシテ テイジスル ヒツヨウ ガ アリマス。 バッグ ヲ オロス トキ ヤ、 コレクション スル トキニ、 ミブン ショウメイショ ノ テイジ ヲ モトメラレル バアイ ガ アリマス。 リョウシュウショ ヲ テイジシナイ ト、 バッグ ガ アナタ ノ モノ デ アル ト サイト ガ カクシンスル マデ、 バッグ ノ ヘンキャク ガ オクレル バアイ ガ アリマス。 ニモツ ガ オクレテ シュウシュウサレル カ、 マッタク シュウシュウサレズ、 コレ ガ リョコウ ノ テハイ ニ モンダイ ヲ ヒキオコス バアイ、 モンダイ ヲ カイケツスル タメニ ハッセイスル ゼン セキニン ト ヒヨウ ハ、 ソノ バショ ガ センデン ヨリ モ ハヤク ヘイササレタ カ、 マタハ シュウシュウ ガ オクレテイタ カ ニヨッテ キマリマス。 ラゲージドロップ マタハ ソノタ ノ ショウゴウ マタハ カイシャ ノ ゲンテイメイ ハ、 オキャクサマ ノ ニモツ ニツイテ イッサイ ノ セキニン ヲ オイマセン。";
            Katakana = KanaConverter.Convert(Katakana, KanaOption.Narrow);

            byte[] convertedBytes = Encoding.Convert(utf8, enc, utf8.GetBytes(Katakana));

            var e = new EPSON();

            printer.Write(
                ByteSplicer.Combine(
                e.CodePage(CodePage.KATAKANA),
                e.PrintLine("")
                )
            );
            printer.Write(e.PrintLine("Katakana: convertedBytes: "));
            printer.Write(convertedBytes);
        `

closed 2020-07-26 by riaanvdl

lukevp, 2020-07-26:
@riaanvdl thanks for the followup with example code! So glad you were able to get it working.

While the translation from English to kanji to katakana is probably not going to be ideal from a literary standpoint, my understanding is that Japanese people are used to using katakana for things that are related to computers because of these limitations, so I would guess that a lot of receipts have half width katakana on them anyway. And indeed, the Wikipedia article on half-width katakana says as much.

```Although often said to be obsolete, the half-width katakana are still used in many systems and encodings. For example, the titles of mini discs can only be entered in ASCII or half-width katakana, and half-width katakana are commonly used in computerized cash register displays, on shop receipts, and Japanese digital television and DVD subtitles```

I imagine this will not be a strange thing to see on a receipt and is the correct way to solve your issue, although the ideal would be for your messaging on the receipt to not originate from computerized English translations but from a native speaker.

If you are ever interested in printing Kanji please let me know and I would be happy to help you add support for bitmap fonts to this library. That would allow printing of any Unicode text, including emojis, provided a fixed size font can be found for generating the bitmaps. I think that would be really cool.