lukevp/ESC-POS-.NET #69: PrintLine() Width

mikart143, 2020-08-17:
Hello!
Thank you for great project.
I am using this project with OKI m3320 printer. All things works great beside one, right alignment. When I print sth with right alignment it cut portion of text (does not fit). Is there any way to set line length ?

lukevp, 2020-08-17:
hey @mikart143 glad you like the library!  As far as I know, you are the first to use a dot matrix printer with the library.  There is no need to specify the line width for EPSON printers, you just call RightAlign() then PrintLine() then CenterAlign().

Are you running it in some type of Epson ESC/POS compatibility mode?

Take a look at the Test Receipt test case in the ConsoleTest app and see how it prints for you.

mikart143, 2020-08-18:
I am using your test receipt :+1: 
Picture: [https://imgur.com/eVXgJ3w](Imgur) (Sorry, but github does not want to embed my photo)
I am aware that esc/pos != esc/p2 but most of the command are the same. I need only to print simple documents.
It seems that printing area is set to whole page including perforated area.
When I applied my printing area (I have no idea if that was correct) the result was that it started printing on the half of the page.(I used some epson document about esc/p2 code from 1997).

Beside all of this. As far as I know there are some differences between esc/pos and esc/p2, mainly in printing bitmaps.
When I will get more familiar with esc/p2 I will may contribute to the project and add some esc/p2 support

mikart143, 2020-08-18:
So, generally this was my fault, I took example to seriously.
When I set ESC Q 75 i wrote 75 as two separate numbers not as single byte :facepalm:. Everything Works Great ! I will test more, When I will get more familiar with this I will try to contribute to the project to add dot matrix printer support. As far as I test, all I need works properly.

closed 2020-08-18 by mikart143