lukevp/ESC-POS-.NET #70: Dot Matrix Printer, unable to print short message

mikart143, 2020-08-18:
Hello!
So far everything works great beside one thing. When I try to print single character I am unable to do this.
```
           printer.Write(new []{(byte) 97}); // Works
           var e = new EPSON();
           printer.Write(
               ByteSplicer.Combine(
                   e.PrintLine("a") // Does not work
               )
           );
```
I inspect your code a little bit and it seems that there is problem with flushing, but I am not sure. You need around 150 characters to be able to print sth.
I do not see any more errors. With this I should be able to make sth like EPSON, but called DotMatrix.

mikart143, 2020-08-18:
So after some investigation I found out that It is good to add some delay before terminating program. Because of the delay the ` e.PrintLine("a")` works perfect. But this behavior is strange. Those two lines are the same!!

lukevp, 2020-11-17:
@mikart143 closing this issue but if you want to tackle this wait method feel free to reopen.

closed 2020-11-17 by lukevp