lukevp/ESC-POS-.NET #71: Exception Printing 2D-Code

gunfg, 2020-08-19:
Hi,

if i want to print a 2D-Code with more than 254 characters i get an exception:

System.ArgumentException: "Code '123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345' is longer than maximum length 254."

Thanks

lukevp, 2020-08-21:
@gunfg this is part of the validation logic for maximum lengths.  If you clone the lib and the test project, you can comment out this validator and see if the QR code properly prints and scans.  If it does, send us a PR with a test case with a QR code longer than 254 characters and increase the validator to a higher amount!

igorocampos, 2020-08-24:
Yeah, this is probably my fault. I didn't have any printers on me when I implemented it, so I used the limit I thought it had. But as @lukevp, once tested that it can work with longer strings, one just needs to change the value in the validator.

gunfg, 2020-08-26:
Ok, thanks.
I can test it when i'am back at office next week.

bazallisonnz, 2020-09-29:
What length is this likely to be changed to? I have had the same issue (QR Code Model2), but the url I am getting back and need to print is over 800 characters. So, at least 1000(?) would suit my requirements. I have tested this locally and it successfully prints the QR Code. :)

igorocampos, 2020-09-29:
Hi @bazallisonnz, I just need to know what's wrong? Is it only the max char, or the qr code types as well? As I mentioned, I don't have a printer anymore, but I am glad to create the PR upon testing that you did. 

Can you confirm that Model1 and 2 are working fine, and just need a higher max char  ?

bazallisonnz, 2020-09-29:
Hi @igorocampos, for me I am only using the Model 2 QR Code setting. I have a local copy of the project and I have only increased the length for the model 2 type to 1024. I can successfully print the QR code which is between 800-900 characters. I don't know enough about the different QR code types to know which ones can reasonably be increased, but I am happy that for my situation I can use the Model 2 with a long url. Thanks. :)

igorocampos, 2020-09-30:
@bazallisonnz would you mind testing for us:
- changing the limit to more than 1024 character (say 2048) and check if it prints correctly as well
- printing the same data in model 1, and see if it's a readable qr code

igorocampos, 2020-09-30:
I couldn't verify ESCPOS/Printer's limitations, but researched QRCode data limitation and opened a PR to fix the validator.

bazallisonnz, 2020-09-30:
OK @igorocampos, I will try this later today after work.

bazallisonnz, 2020-10-01:
Hi @igorocampos, I have tested using the Epson TM T88 and the Console Tests:

Print2DCode(TwoDimensionCodeType.**QRCODE_MODEL2**, websiteString):
I was able to push this out to 2048 characters, but this is marginal in terms of sizing.

Print2DCode(TwoDimensionCodeType.**QRCODE_MODEL1**, websiteString):
After a bit of trial and error I was able to get a QR Code at 675 characters, but not at 700. Sorry, I don't feel like more testing to eek out any more characters than that. :)

igorocampos, 2020-10-01:
Got it, Model 1 number is pretty close to the new limits I've put in the PR for @lukevp

Maybe aside from the qr code limit itself there must be another underlying printer limit. And maybe that vary from printer to printer.

Let's wait on Luke's opinion about it and if he wishes I can change the PR accordingly

bazallisonnz, 2020-10-01:
Hi @lukevp, I am not sure what values @igorocampos has provided, but just with respect to Model 2, I would like a limit of (around / at least)1024 which the Epson TM T88 happily printed (it even printed 2048, but...). I currently get a link provided which is between 800-900 characters, so ~1000 would provide some headroom. Thanks for this solution by the way. :)

lukevp, 2020-10-01:
Yeah, for QR codes the block size will get to a point where it will not be precise enough for the error correction to overcome the printer quality for specific sizes.  This would be affected by how good your thermal paper is, how smooth the rollers, and how new the print head, so I don't think the library should really enforce "practical" limits as far as maximum precision of the physical hardware, just limits that are at the protocol/data specification level of QR codes themselves.

I'll see how far I can get just for fun, I have a very new printer and a really accurate/high end physical barcode scanner, but I think we should go with the documented max limits in case someone is able to print them better than us in the future.

bazallisonnz, 2020-11-11:
Hi @lukevp, is there any timeframe where we might see a new release with this enhancement? Thanks.

igorocampos, 2020-11-11:
Just as a recap, the PR is #79

lukevp, 2020-11-12:
Thanks for the ping, I will do a release this weekend if I can.  If I don't close this out by Monday feel free to ping me again.

lukevp, 2020-11-16:
Will be preparing this release this evening!

lukevp, 2020-11-17:
@bazallisonnz 1.5.0 released @ https://www.nuget.org/packages/ESCPOS_NET/1.5.0
PRs #79 and #84 included along with some improvements from #82 . Thanks again for the ping and sorry for the delay.  Please close this out if the new version works for you!

bazallisonnz, 2020-12-01:
Thank you for that @lukevp, I have tested my scenario and I am very happy that it now prints the somewhat long url I get as a QR Code (Model2). I really appreciate the time and effort you put into this! And thanks @igorocampos for creating the pull request. For me, I see this as resolved. :)