lukevp/ESC-POS-.NET #72: How to save and reuse print buffer?

Spoorthy283, 2020-08-23:
Hi, I am able to write a receipt to the printer as shown in the wiki page (Step 2). I would like to save the print buffer and reprint the print buffer again. Can you please let me know if this is possible through your library. If yes, can you please give a small illustration.

Usecase for this requirement:
1. Reprint the receipt if the customer comes back with a request for duplicate copy
2. Incase of returns print two copies of the receipt. Second copy with signature lines so that customer can sign.

lukevp, 2020-08-24:
Once you create an emitter, all it's really doing is generating byte arrays.  You can use the output of the ByteSplicer.Combine(...) or combine the byte arrays however you choose.  That's all you need to store to be able to reprint (this is assuming you are generating receipts for the same printer as you want to print them against in the future.)

You can store this byte array however you are persisting data for your app.  Write to a file, save to a blob or a varbinary in a database, store in Redis, whatever you choose that can store binary data.  Then when you need to reprint, you just call 
printer.Print(byteArray);

This will work across printer types (eg. you can print a receipt to a combination of a network printer and a USB printer if you choose).

If you want to print 2 copies, one with a different footer, you can just store the byte array in memory, for example

```csharp
var receipt = ByteSplicer.Combine(e.PrintLine(), e.PrintLine("Hello"), e.PrintLine());
printer.Print(receipt);
var copyOfReceipt = ByteSplicer.Combine(receipt, e.PrintLine("Signature: ____________");
printer.Print(receipt);
```

Please let me know if that clarifies things for you, and if you have suggestions on how this could be clarified for others in the readme!

Spoorthy283, 2020-08-24:
Hi, Thanks a lot for the quick and detailed explanation. I was able to save the print buffer, retrieve it and print it again. I was facing lot of challenges with printing on linux. This Nuget is simply awesome as it does not need any printer drivers. This has almost everything that a receipt needs. You guys are doing an awesome job. Thanks again.

A quick question, 
 e.SetStyles(PrintStyle.Bold) worked perfectly fine but e.SetStyles(PrintStyle.Underline) did not work. Not sure if it has to do with the printer of the library. Any idea what could be wrong?

varunbhoopalam3, 2020-08-24:
Hi Luke, I had a similar requirement and I was able to save the byte array and reuse it based the method you suggested above. I also have a requirement to preview the print buffer before reprinting it. Is there a way I can convert the byte array to an Image or may be UTF-8 text so that I can display the preview to the user before they decide to print? Thanks a lot for your help.

lukevp, 2020-08-24:
@varunbhoopalam3  I'm glad the byte array storage worked for you!   Creating a system to create an image of the buffer is something I'd love to incorporate into the library.

ESC/POS is a binary protocol, and the library currently outputs byte array strings for interpretation by the Epson printer.  To generate an image or a UTF-8 of the text would require creating a printer emulator / simulator in code that could read an incoming byte array and render an image or string output.  I'm not opposed to this at all and would be happy to help, but it is not trivial.

Here is an example of the process by the person behind escpos-php.
https://mike42.me/blog/2018-05-recovering-text-from-a-receipt-with-escpos-tools

lukevp, 2020-08-24:
@Spoorthy283 Thank you for the kind words!  I'm glad that everyone has been enjoying this library so much, and I'm excited to see it continue to grow.  

Which printer are you using?  I would suggest cloning the repo and executing the ConsoleTest app, and run through all the test cases (QR codes, styles, barcodes, images, etc.) if anything fails to print, you may either have a printer that doesn't fully support ESC/POS, or you may have a driver/interface related issue with your printer.