lukevp/ESC-POS-.NET #74: Unreadable 2D code

neon-dev, 2020-08-28:
I've noticed that QRCode Model 1 is not readable. And Model 2 looks to be Model 1 when visually comparing the printed code with a Model 1 code printed via this pre ESC-POS implementation:
```cs
String text = "test";

printerBuffer[0] = 0x1d;
printerBuffer[1] = 0x28;
printerBuffer[2] = 0x6b;
printerBuffer[3] = 4;
printerBuffer[4] = 0;
printerBuffer[5] = 0x31;
printerBuffer[6] = 0x41;
printerBuffer[7] = 0x49; // 0x49 - model 1 / 0x50 - model 2
printerBuffer[8] = 0x00;
serialPort.Write(printerBuffer, 0, 9);

printerBuffer[0] = 0x1d;
printerBuffer[1] = 0x28;
printerBuffer[2] = 0x6b;
printerBuffer[3] = 3;
printerBuffer[4] = 0;
printerBuffer[5] = 0x31;
printerBuffer[6] = 0x45;
printerBuffer[7] = 0x33; // 0x33 - Error correction level H (30% recovery capacity)
serialPort.Write(printerBuffer, 0, 8);

int len = 3 + text.Length;

printerBuffer[0] = 0x1d;
printerBuffer[1] = 0x28;
printerBuffer[2] = 0x6b;
printerBuffer[3] = (byte)(len % 256);
printerBuffer[4] = (byte)(len / 256);
printerBuffer[5] = 0x31;
printerBuffer[6] = 0x50;
printerBuffer[7] = 0x30;

char[] strChars = text.ToCharArray();
for (int i = 0; i < text.Length; i++) {
    printerBuffer[8 + i] = (byte)strChars[i];
}
serialPort.Write(printerBuffer, 0, 8 + text.Length);

printerBuffer[0] = 0x1d;
printerBuffer[1] = 0x28;
printerBuffer[2] = 0x6b;
printerBuffer[3] = 3;
printerBuffer[4] = 0;
printerBuffer[5] = 0x31;
printerBuffer[6] = 0x51;
printerBuffer[7] = 0x30;
serialPort.Write(printerBuffer, 0, 8);
```
This is the equivalent ESC-POS call:
```cs
printer.Write(E.Print2DCode(TwoDimensionCodeType.QRCODE_MODEL2, text, Size2DCode.SMALL, CorrectionLevel2DCode.PERCENT_30));
```

igorocampos, 2020-08-28:
Hello there, according to this enum QRCODE_MODEL2 should be cast as `0x50`, but you are saying it results as `0x49` ?

[image]

Can you debug and provide the final byte array values returned by  `E.Print2DCode` in your example? So we could compare byte by byte to your old code example?

lukevp, 2020-08-28:
@igorocampos  I think this enum (and possibly the others) is set wrong, if it should be hex 49 it needs to be specified as 0x49, not 49.  49 would be decimal 49 = hex 0x31. WHat do you think?

igorocampos, 2020-08-28:
It is possible. I checked out the manual I had, and it is not clear if it means decimal or hexa. But look how the function number in the title it uses decimal: `fn=65` and not the hexa `0x41`. So I assumed it was a pattern of the manual to describe all in Decimals

[image]

neon-dev, 2020-08-28:
Don't trust this legacy code I posted too much. It was never used in production and I just inherited the project, so no guarantees for correctness. Now that you say it, to me it looks like the byte stream is incorrect and 0x49 should in fact be 49 / 0x31, since all other codes for commands I've seen were decimals.

I cannot test anything right now, so the only thing I can assure is that codes with QRCODE_MODEL1 failed to scan but QRCODE_MODEL2 can be scanned successfully. Whoever else has access to a printer should verify first.

igorocampos, 2020-09-30:
Please read some more about this in #79

neon-dev, 2021-04-26:
Hi, I finally had the chance to test a bit on a TM-T88IV printer. Using values `49`, `0x49` and `0x50` resulted all in model 1 QR codes. Only `50` created a model 2 code, which I identified by the alignment pattern visible in the bottom right corner on size 4 (medium). So decimal values are correct.
However I still could not ever read any of the printed model 1 codes. Maybe it's an issue with the scanner apps I tried.
If anyone can tell me an app that is known to scan model 1 codes I'll gladly try and if it succeeds we can close this issue.
Thanks