lukevp/ESC-POS-.NET #77: Can't print special characters ?

NgyAnthony, 2020-09-25:
Hi, I'm having a little bit of an issue with printing accents and I would like your help.

### Informations:
Printer: Epson TM-20III
Os: Windows 10

### Here's the problem:
- Printing any special character (é, ç, ë, ...) will give another special character (é becomes Ú for example).

### Heres what I tried: 
- I took a look at #67 and tried multiple `CodePage` but I still get the correct characters.

- I also tried to identify which encoding was returned by `e.PrintLine("é")` so that I could maybe convert it to the correct encoding but I couldn't identify which encoding was used in the returned byte[].

- I modified my CodePage and language from US to French from the EPSON Utility but nothing changed.

- I successfully printed a test page of the codepage, including special characters.

At this point I'm kind of lost. I could parse all the special characters out of my string but I hope that I don't have to do that...

If you could give me some information it would be greatly appreciated 🙏

NgyAnthony, 2020-10-07:
Couldn't find a solution, I'm removing all accents in the strings I'm printing.
https://stackoverflow.com/questions/249087/how-do-i-remove-diacritics-accents-from-a-string-in-net

lukevp, 2020-10-07:
Hey @NgyAnthony, sorry I missed your original question. 

The ESC-POS standard is not Unicode-Aware, and can only print ASCII characters.  To support characters with diacritics, the code pages allow reassigning some extended (non-latin) areas and/or CJK via Katakana.  So for example a 0x30 printed with one code page may be one letter and with another code page a different letter.  That is probably why you are seeing different diacritics or characters printed when you change the code page.

However, C# has Unicode strings by default, so you will need to convert these Unicode strings into ASCII strings under a specific code page FIRST, and then ALSO set the Epson printer to this same code page.  Unicode is a 16-bit space (so imagine there are 65,536 unique letters possible in Unicode).  ASCII is 7-Bit, so there are only 127 possible values.  Code pages allow you to use these 127 values to represent different strings, based upon the intended character set, but you will need to know both at the time you convert from Unicode to ASCII and from when you send an ASCII byte string to the printer which code page you are leveraging, and they need to match.

Hopefully that is helpful, but please let me know if it's unclear and I'll try to provide more detail.

NgyAnthony, 2020-10-08:
Hi, thanks for your explanation, it was very clear.

I managed to be able to print accents with this snippet:

```
        public void PrintTest()
        {
            string unicodeString = "ééééééàçàçàç";
            byte[] tempBytes = Encoding.GetEncoding("IBM00858").GetBytes(unicodeString);
            
            printer.Write(
              ByteSplicer.Combine(
                e.CodePage(CodePage.PC858_EURO),
                tempBytes,   // Works
                e.Print(Encoding.GetEncoding("IBM00858").GetString(tempBytes)), // Doesn't work
                e.Print(Encoding.ASCII.GetString(tempBytes)), // Doesn't work
                e.PrintLine("Test"),
                e.FullCutAfterFeed(30)
              )
            );
        }
```

However I'm unable to use the library's functions (e.Print or e.PrintLine), I have to give the ByteSplicer my byte array directly. Any idea on how I could convert my byte array to a string that is accepted by e.PrintLine ?

Thanks again for the input !

NgyAnthony, 2020-10-09:
Closing the issue since I found a fix.

closed 2020-10-09 by NgyAnthony

rsantosdev, 2021-09-08:
Thanks, @NgyAnthony for posting this. This was the solution to make the euro symbol and the British pound symbol work as expected.

Just some small comments if anyone has the same issue:

1 - You need to register the **IBM00858** encoding:
```
// On your Program.cs add this line
Encoding.RegisterProvider(CodePagesEncodingProvider.Instance);
```