lukevp/ESC-POS-.NET #82: Add Fluent API to improve Developer Experience

lukevp, 2020-10-01:
I'd like to do that but the Print method of the emitter is responsible for sanitizing strings and fixing newlines in the strings, so you could do 
```csharp
ByteSplicer.Combine(e.CenterAlign(), e.Print("A CENTERED TEXT"), e.LeftAlign(), e.Print("LEFT ALIGNED TEXT"));
```
but the byte splicer does not itself know what emitter you're using, so we can't assume to know how to convert the text to proper print commands in the future if they are not able to be emitted directly as text (say for a different type of thermal printer).  

Perhaps this could be a good use case for a fluent API like 
```csharp
var DirectPrinter = new DirectPrinter(emitter, printer);
DirectPrinter.CenterAlign()
             .Print("Centered Text")
             .LeftAlign()
             .Print("Left Aligned Text")
             .Send();
```

and the .Send() method would internally maintain a byte array and call bytesplicer.combine and send on to the underlying printer whenever it's desired?  This would work more like some other people's libraries do but still have the composability of different emitter and printer types.  The downside is the DirectPrinter would have to wrap every single command with the same signatures as the underlying emitter.

_Originally posted by @lukevp in https://github.com/lukevp/ESC-POS-.NET/issues/53#issuecomment-638443847_

igorocampos, 2020-10-01:
Totally forgot we had this conversation, lol

I think with DirectPrinter we can allow both ways (mostly for retro compatibility), but instead .Send() method maintaining a byte array, I'd rather have the DirectPrinter class have it, since it's not a static class anyway.

The mentioned downside is going to be there still, but the other option would be move the clean up from emitters? Don't think it's better than having DirectPrinter wrap every command.

lukevp, 2020-11-17:
Added a virtual Write method to BasePrinter that takes a byte array array, so you don't have to call ByteSplicer.Combine() for everything, you can just pass byte array arrays to Write directly and it will call ByteSplicer.Combine internally.

PR #136 (2021-10-02): Creation of new Fluent API through DirectPrinter class
Fixes #82 

@lukevp, I finally got back to this. 

The end result is exactly what you suggested in #82 

```cs
var DirectPrinter = new DirectPrinter(emitter, printer);
DirectPrinter.CenterAlign()
             .Print("Centered Text")
             .LeftAlign()
             .Print("Left Aligned Text")
             .Send();
```

There's an internal buffer that holds the byte arrays being added. Once the `Send` method is called, the buffer is sent to `printer.Write` and the buffer is cleared out. This means that the code below will send "ABC" t [... cut]
closed 2021-10-02 by lukevp