lukevp/ESC-POS-.NET #83: Unit tests

igorocampos, 2020-10-01:
> I would love some unit tests on the project, I didn't need them at the time because I was interactively testing on the printer, so you can consider the ConsoleTest as an integration test suite. The main issue with the unit test design will be what is under test, and how we can properly assert failures. Should we test for specific byte strings on the emitters? Which sections should have test cases built first? Why don't you open an issue where we can discuss the unit tests and add them to the project/wiki for coordination?

From Luke in [here](https://github.com/lukevp/ESC-POS-.NET/pull/81#pullrequestreview-500691111)

igorocampos, 2020-10-01:
>Should we test for specific byte strings on the emitters? 

I think so.

>Which sections should have test cases built first?

I think we can start testing methods that don't need a byte array to be asserted. Like the one I mentioned in #81

lukevp, 2020-10-02:
I'm good with byte array assertions, we just need to make sure we structure the test project so that there is a tester per emitter (right now we just have the EPSON emitter but in the future there will be others).  Agreed that we can do the other unit tests first.  Do you want to make a PR to add a test project as well as a first test so we can start on this? I typically use xunit, is that OK with you?

Unrelated to this ticket, but I don't have good contact info for you @igorocampos.  I'd love to talk to you more about your background and your interests, and see if there's opportunity for us to work together on some other projects.  You can check my website for my [contact information](https://lukep.dev) and send me an e-mail if so.

igorocampos, 2020-10-02:
Yeah, sounds good I'll come up with the PR. xUnit is fine, I actually prefer it over MSTests.

I've sent you an email, please reach out whenever is good :)

PR #84 (2020-11-17): Unit tests
Fixes #83 

@lukevp  I think this is a start, please check it out so a good pattern can be established.
closed 2020-11-17 by lukevp