lukevp/ESC-POS-.NET #86: GB18030 simplified chinese

koo9, 2020-10-29:
did anybody manage to print chinese simplified character with this lib? if so, any code example please. thx

lukevp, 2020-10-29:
Hey @koo9 as far as I know, the only asian language printed so far is katakana.  Simplified Chinese likely needs to be enabled, but it's pretty straightforward to add new command sets to the library.  Please feel free to open a PR! [Check out this link for a similar issue](https://github.com/qzind/tray/issues/285)

koo9, 2020-10-29:
yes, it's a matter of adding codepage to the codepage enum. or add an extension method for that. I am using xprinter. it supports GB18030 encoding.

koo9, 2020-10-29:
Ok found a solution for that. Im using Xprinter which supports GB18030 Simplified Chinese characters. so by default it's set to that codepage. in order to print chinese character, we need to encode the string properly like so:

`string text = "中文菜单";
            var encoding = System.Text.Encoding.GetEncoding("GB18030");
            byte[] bytes = encoding.GetBytes(text);
            string isoString = Encoding.GetEncoding("ISO-8859-1").GetString(bytes);

            var e = new EPSON();
            printer.Write(ByteSplicer.Combine(
                e.Initialize(),
                //e.CodePage(CodePage.CHINA),
                e.PrintLine(isoString),
                e.FullCut()
            ));`

No need to set the codepage here and I don't think the current implementation have the codepage for that. the key here is to convert the string inot ISO-8859-1 encoded string.

Hope that helps others.

closed 2020-10-29 by koo9

xklake, 2023-01-30:
> yes, it's a matter of adding codepage to the codepage enum. or add an extension method for that. I am using xprinter. it supports GB18030 encoding.

i am also using xprinter N160i wifi printer, but this GB18030 is not found and throws an exception,  does model relate or code issue? 

Thanks