lukevp/ESC-POS-.NET #87: Example of using the library in Xamarin ?

kwiksys, 2020-11-02:
Appreciate is there an example or documentation on applying the library in Xamarin ?

Regards

lukevp, 2020-11-29:
hi @kwiksys @fkbeys this is a .net standard 2.0 library, so it should work in Xamarin with the examples given, but depending on the platform used, you may not be able to communicate with all types of connectivity (for example, on Android, you should be able to use the NetworkPrinter to print over WiFi, or use the Serial interface to print over a virtual serial port on Bluetooth, but you may not be able to use the FilePrinter or the USB Printer).

lukevp, 2020-11-29:
see #66 as well, several people have asked for a Xamarin example so this would be a great thing to PR if you want to take a stab at it! I'd be happy to help wherever I can.

fkbeys, 2020-11-29:
@lukevp 
I just need to see an example of a code

How to use Serial interface to print over a virtual serial port on Bluetooth?

igorocampos, 2020-12-17:
@fkbeys please check #51 for a code snippet of what you asked.

fkbeys, 2020-12-24:
> @fkbeys please check #51 for a code snippet of what you asked.

I have already implemented the text printing.Its working. 
I wonder to know, how can i print a little image? 
I tried to convert the image to byte array, but my printer prints random numbers and characters. 
How can i solve this issue? thank u

Sakab, 2021-06-15:
> I have already implemented the text printing.Its working.
> I wonder to know, how can i print a little image?
> I tried to convert the image to byte array, but my printer prints random numbers and [...]

It depends from the printer itself. Sometimes printer can't handle the byte array, printing a "interpretation" of that in characters (letters, numbers and symbols). That happend to me many times, especially if printer use another language beside ESC-POS. Maybe lowering resolution or use the "IsLegacy" overload can help.

lukevp, 2021-09-24:
@fatihbabacan92 this is the issue I was talking about, along with #66.  Would be great if you could open a PR with an example app for Xamarin, like the ConsoleTest app, and we can have an example of printing cross-platform from mobile :)

closed 2021-09-24 by kwiksys