lukevp/ESC-POS-.NET #88: How to set the charsetcode? when I try to print chinese,it's show Garbled code

alexzm19, 2020-11-10:


lukevp, 2020-11-11:
hey @alexzm19 there is not currently a way to print unicode characters, so you need to use a charset that's built in to the printer.  Take a look at these issues for more information: #86 #67 . Give it a try and let me know how it goes!

alexzm19, 2020-12-15:
> hey @alexzm19 there is not currently a way to print unicode characters, so you need to use a charset that's built in to the printer. Take a look at these issues for more information: #86 #67 . Give it a try and let me know how it goes!

thank you for answer,I solved the problem
step1: In .net core 3.1,.NET Core does not support GB2312 by default
step2:Register encodingProvider on startup
step3:Change print function
return data.ToCharArray().Select(x => (byte)x).ToArray()
=>
return Encoding.GetEncoding("GBK").GetBytes(data)

lukevp, 2023-02-02:
Glad we got you solved!  @igorocampos this is a common question we get about printing nonstandard character codes.  Perhaps we should put something about this in the README?

closing for staleness.

closed 2023-02-02 by lukevp

igorocampos, 2023-02-09:
Yeah, sounds about right. In addition I do think we could make the Enconding changeable instead of hardcoded UTF-8. See the comment above, Alex had to change the Print method in order to place his own encoding in it. I think we can keep UTF-8 as default, but also let the user change it, if wanted.

I'll see if I can put a PR together about that. 

Btw, I'm in the US now :)

igorocampos, 2024-01-23:
@alexzm19 Can you describe with code what `step2:Register encodingProvider on startup` means?
I wish to add this steps in the README.md in details.  Thank you

alexzm19, 2024-01-23:
of course
Code:
Encoding.RegisterProvider(CodePagesEncodingProvider.Instance);
Link:
Encoding.RegisterProvider(EncodingProvider) 方法 (System.Text) | Microsoft Learn

rainonheaert
***@***.***

igorocampos, 2024-01-23:
@alexzm19 thanks! I see you replied by email, but I think the link got lost when doing so. When you have the time, if could come back to Github and update your comment adding the URL for the link you mentioned, I'd appreciate it!

alexzm19, 2024-01-23:
Ok, I just updated the comment, thanks for the reminder

rainonheaert
***@***.***

igorocampos, 2024-01-23:
Thank you very much! That will do :)
Here's some more from the doc:
[image]