lukevp/ESC-POS-.NET #89: Throws exception(System.IO.FileNotFoundException) when used in .net core 3.1

ahoirg, 2020-11-10:
I added ESCPOS_NET from nuget to a project created using .net core 3.1. ESCPOS NET cannot load. throws the following error:

- $exception {"Could not load file or assembly 'ESCPOS_NET, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.":"ESCPOS_NET, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null"} System.IO.FileNotFoundException

lukevp, 2020-11-10:
hey @ahoirg, 

How did you add it to the project?  ESCPOS_NET targets .net standard 2.0 and should work on this version of .net core.

What platform are you on?

ahoirg, 2020-11-10:
I'm using .net core 3.1.
 .net standard 2.0 only works on .net core 2.0. Will you switch to .net standard 2.1 to work on .net core3.0 or 3.1?

lukevp, 2020-11-10:
.net standard 2.0 works on all .net core from 2.0 and up.  from [this link](https://docs.microsoft.com/en-us/dotnet/standard/net-standard):

> The following table lists the minimum implementation versions that support each .NET Standard version. That means that later versions of a listed implementation also support the corresponding .NET Standard version. For example, .NET Core 2.1 and later versions support .NET Standard 2.0 and earlier versions.

ahoirg, 2020-11-11:
The methods I tried are as follows;
1) I installed SC-POS-.NET from Nuget. I imported the class I will use.I got an error.
2) I downloaded the repos. I switched the project to .net standart 2.1.  I imported the class I will use.I got an error.

screenshot of the error i got in both cases:
[image]
[image]

lukevp, 2020-11-11:
Something is wrong with your environment or your build configuration (sounds like it's not publishing or building the dependencies correctly).  This is not a compatibility issue with .net standard or with this library.  It is building with a reference to ESCPOS_NET DLL, but at runtime, it can't find the assembly.  If you check in your output / publish directory, you should see that the DLL is missing.

Try making a brand new console app in .net core 3.1 and reference ESCPOS_NET and it will work fine.  I did this to double-check and it had no issues building, importing the package, or creating a class from the assembly.

I can help you troubleshoot if you're willing to put the code on GitHub.

[ConsoleApp3.zip](https://github.com/lukevp/ESC-POS-.NET/files/5524805/ConsoleApp3.zip)

ahoirg, 2020-11-11:
The project I am working on is not open source and unfortunately I am not authorized to share it.

 It worked in my solution, when I added the dll to the console application. I also looked at the project you zipped. As far as I understand, when I add it to a console application, the dll works without any problems.
[image]

but when I added it in a class library type project, the dll did not work.
[image]

I created an example solution:

[TempESCPOS.zip](https://github.com/lukevp/ESC-POS-.NET/files/5527189/TempESCPOS.zip)

lukevp, 2020-11-12:
You can't directly start a class library, how are you referencing the class library?  

Is the problem that you have the library incorporated into a class library, then you have another app that references the class library? What type of app is that?

I downloaded your example and I can start the ConsoleApp1 with no issues, but I cannot start the deneme project, because it is a class library.

I am on Windows 10 2004 with Visual Studio 2019 Community, what are you using?

ahoirg, 2020-11-12:
Actually I have plugins [application](https://docs.microsoft.com/en-us/dotnet/core/tutorials/creating-app-with-plugin-support). Naturally, there are many projects in this solution.

When library type projects are compiled, they create dlls in a common area. The application runs from a winexe type project. Then the application looks at the common folder and loads the dlls. It decides which plug-in will work according to the requests and performs the operations.

Library type projects can refer to other library type projects. When compiled, it creates the dll of these projects. Or if it uses a specific namespace, it creates the dll of the namespace it uses. For example INIFileParser.dll

I am on Windows 10 Pro. Also i am using Visual Studio Professional 2019.

closed 2020-11-13 by ahoirg

lukevp, 2020-11-16:
Thanks for following up and closing out the ticket.  I do think this is not related to ESCPOS_NET and the issue should be closed, but I'm happy to continue helping you with your project if you have any more questions.  

It sounds like the problem is in the AssemblyDependencyResolver, where it should be resolving the dependency of the plugin that references ESCPOS_NET but it 

1) isn't resolving dependencies at all (implementation problem), 
2) isn't evaluating the dependency graph properly (doesn't see ESCPOS_NET as a dependency)
or
3) can't find the dependencies because of some pathing issue.  

You might try loading the base assembly with AssemblyLoadContext and then check if you can introspect it in the debugger and see if its dependency graph is set up right.  I haven't ever done this, but it seems like it should be possible.

ahoirg, 2020-11-16:
I had same problem in other 3rd party DLL. The problem seems to be with the AssemblyDependencyResolver structure.