lukevp/ESC-POS-.NET #92: Make method virtual on BaseCommandEmitter

danielmeza, 2020-11-29:
Make all method virtual on BaseCommandEmitter do that can be overrided on each printer implementation of needed. 

This surge because I was testing the a Custom printer VKP80ii and the cut command is not responding, the work when I use the Clear command. Everything else is working properly except the paper cut.

Wit this we can create a new class for that printer override the FullCut method and then fix the issue without repeate code.

igorocampos, 2020-12-17:
The more proper way is actually to contribute with the library and implement those different/specific emitters like VKP that you mentioned.  Right now there's only EPSON's one.

But I think I get your point. You want to be able to modify the behavior of an emitter without the need to change the library itself. Is that it @danielmeza ?

What do you think about it @lukevp ?

danielmeza, 2020-12-18:
Yes! @igorocampos  the VKP printer emulate ESC POS bute with some differences, the idea is to recycle the emitter an only implement those differences.

lukevp, 2020-12-18:
Yes @danielmeza  that’s the idea I had, that you could derive a class from the base emitter and override anything that’s weird / custom or an extension of the normal escpos functionality. Could you open a PR with the changes you’d like so we can comment and review there?

igorocampos, 2020-12-24:
@lukevp I think @danielmeza was right, I went to check the methods and none were virtual (so no override was possible). I've created #98 to solve that, please check it out.

jmalda, 2021-02-02:
Hi @danielmeza @igorocampos  . Making all methods virtual on BaseCommandEmitter is a great idea. I am working with CUSTOM Printers as well (specifically the KPM180H and TPTCM112II). CUSTOM follows the  EPSON ESC \ POS protocol fairly closely, but as you said, there are a few commands they have implemented in a slightly different manner. A case in point is printing QR Codes.

For EPSON, the Barcode class has this:
```
public static readonly byte[] StoreQRCodeData = { 0x31, 0x50, 0x30 };
public static readonly byte[] PrintQRCode = { 0x03, 0x00, 0x31, 0x51, 0x30 };
```
while for CUSTOM,  the commands need to be as follows:
```
public static readonly byte[] StoreQRCodeData = { 0x31, 0x50, 0x31 };
public static readonly byte[] PrintQRCode = { 0x03, 0x00, 0x31, 0x51, 0x31 };
```

Making the methods virtual would allow the BaseCommandEmmitter defaults to be overridden for specific implementations.

igorocampos, 2021-02-02:
@jmalda the PR to change it is done (#98), we just need @lukevp approval on that

danielmeza, 2021-02-02:
@lukevp are you there? we need this PR merged!

danielmeza, 2021-02-02:
@jmalda Thanks for the CUSTOM command for the QR Codes!

jmalda, 2021-02-03:
@danielmeza You are welcome! Currently the library just has an EPSON implementation. I hope to create a PR in the next week or two that has the beginnings of a CUSTOM implementation that overrides a number of the commands that are slightly different for CUSTOM's thermal printers such as the TPTCM112II and KPM180H. It could be added to over time as we come across other commands that have been implemented slightly differently by CUSTOM.

closed 2021-02-03 by lukevp

lukevp, 2021-02-03:
@danielmeza @igorocampos @jmalda  thanks for the discussion and the PR! This was my intent with the base command emitter originally, but I don’t have any non epson printers so I never implemented a second emitter. PR is merged but if we need to reopen this issue once you implement the CUSTOM commands, let me know. Please feel free to open a PR with the new changes.

PR #98 (2021-02-03): Virtual methods in BaseCommandEmitter
fixes #92