lukevp/ESC-POS-.NET #94: Line Display Sample Code

dicksonkimeu, 2020-12-19:
Any sample code for line display peripherals?

lukevp, 2020-12-19:
Hey @dicksonkimeu !
I was working with someone who had a line display and a cash drawer a while back but they never tested the line display. Check out #17 for some background. The main commands you need are initialize, clear and print. The scrolling commands can all be implemented on top of those primitive operations (but if you’re interested in using those directly, let’s talk about it!) I don’t have a working line display so I’m not able to help test/implement much, but if you need this for work and have an extra line display you can ship me, I can certainly assist.

dicksonkimeu, 2020-12-19:
What's your email?

lukevp, 2020-12-19:
You can reach me at luke.paireepinart. It’s hosted through Gmail.

dicksonkimeu, 2020-12-22:
Hi, I dropped you an email .

dicksonkimeu, 2020-12-25:
Hello, I have a line display and ready to test

lukevp, 2020-12-25:
Hi @dicksonkimeu ,
did my previous reply not give you enough info to get started? It should be pretty straightforward!

dicksonkimeu, 2020-12-25:
@lukevp seems am missing something, could you share the sample code I can run to test. The current implementation is enough for most line displays.

Also I got a question; 
In some setups especially restaurants the line display is a screen that accommodate more content. Would the current implementation support scrolling ?

lukevp, 2021-02-15:
The line display just renders whatever is printed to it, so you can implement scrolling in C# using a clear command and truncating the string (use .Skip() Linq method, for example).   This is more flexible than the scrolling built into the line display but would require continually sending text to the line display.

I can't provide sample code that works, since I don't have a line display, but you basically just say you're printing to the line display then write to the stream.

closed 2021-09-05 by lukevp