lukevp/ESC-POS-.NET #95: Support for Fiscal Printer

dicksonkimeu, 2020-12-19:
Does this support fiscal printer ?

lukevp, 2020-12-19:
I’m not sure what a fiscal printer is, could you provide a reference to this please?

dicksonkimeu, 2020-12-19:
https://docs.microsoft.com/en-us/previous-versions/windows/embedded/ms884287(v=winembedded.4)

dicksonkimeu, 2020-12-20:
Hi @lukevp ,

A thermal receipt you pass a free text to the sdk. A thermal fiscal receipt there are certain commands that should be passed to the SDK.

It's still ESC/POS commands. How can I pass ESC/POS commands using the SDK ?

LIST OF FISCAL COMMANDS - IN ASCENDING ORDER

HEX	DEC	Function
21h	(33)	Clear the display
23h	(35)	Show text on lower line of display
26h	(38)	Open non-fiscal receipt
27h	(39)	Close non-fiscal receipt
29h	(41)	Setting the memory switches
2Ah	(42)	Printing non-fiscal free text
2Bh	(43)	Set FOOTER and printing options
2Ch	(44)	Advance paper
2Dh	(45)	Paper cut
2Eh	(46)	Set HEADER (Name and address)
2Fh	(47)	Showing text on upper line of display
30h	(48)	Open fiscal receipt
31h	(49)	Register sale
32h	(50)	Tax rates set during selected period
33h	(51)	Subtotal
34h	(52)	Register sale and show on display
35h	(53)	Calculate TOTAL
36h	(54)	Print free fiscal text
38h	(56)	Close fiscal receipt
3Ch	(60)	Cancel fiscal receipt
3Dh	(61)	Set date and hour
3Eh	(62)	Get current date and hour
3Fh	(63)	Show date and hour on display
40h	(64)	Info on last fiscal entry
41h	(65)	Info on daily totals
43h	(67)	Info on daily paid sums
44h	(68)	Number of free fields in fiscal memory
45h	(69)	Daily financial report with/without closure
46h	(70)	Internal debiting/crediting
47h	(71)	Print diagnostic info
48h	(72)	Fiscalization
49h	(73)	Detailed report of the fiscal memory selected by number of entry
4Ah	(74)	Read statuses
4Ch	(76)	Status of the fiscal transaction
4Fh	(79)	Short report of the fiscal memory selected by date of entry
50h	(80)	Sound signal
53h	(83)	Set multiplier, decimals, currency name and disabled taxes
54h	(84)	Print a bar code
55h	(85)	Set additional payment names
56h	(86)	Get last fiscal memory date
59h	(89)	Program production test area
5Ah	(90)	Return diagnostic info
5Bh	(91)	Program serial number, country number and Fiscal memory number
5Eh	(94)	Detailed of fiscal memory (selected by date of entry)
5Fh	(95)	Short report of fiscal memory (selected by entry number)
60h	(96)	Set tax office text
61h	(97)	Return tax rates
62h	(98)	Set tax registration number
63h	(99)	Return set tax registration number
64h	(100)	Show free text on display
65h	(101)	Set operator’s password
66h	(102)	Enter operator’s name
67h	(103)	Info on current receipt
69h	(105)	Operator report
6Ah	(106)	Drawer kick-out
6Bh	(107)	Define items and items info
6Ch	(108)	Detailed daily report
6Dh	(109)	Print duplicate receipt
6Eh	(110)	Additional daily info
6Fh	(111)	Report on groups of items
70h	(112)	Reading info on operator
71h	(113)	Read the number of the last fiscal entry or period
72h	(114)	Read info on fiscal entry or period
73h	(115)	Program graphic logo
74h	(116)	Read fiscal memory block
76h	(118)	Register technical intervention
78h	(120)	Electronic journal support
79H	(121)	Read code memory (firmware)
7Eh	(126)	Erase electronic journal
7Fh	(127)	RAM reset

lukevp, 2020-12-20:
Hey @dicksonkimeu ,

this is not currently supported, but would be implemented as a FiscalEmitter. We currently only have the Epson emitter. You would not need to use that Fiscal SDK as that’s a different interface (UPOS) vs directly interacting with the printer like this library.

are you open to implementing the Fiscal support? I will assist you as I can but I do not have any hardware that works with this, so we would have to collaborate on a PR.

igorocampos, 2020-12-22:
I've worked with Fiscal Printers here in Brazil 10 years ago (or so), however, I did not use direct command bytes, but rather each manufacture DLL. You will need the whole command to implement it. @dicksonkimeu, the list that you provided is only a summary. You will probably need the parameters info as well, if you provide a handbook I can also assist to implement the commands in a new emitter.

dicksonkimeu, 2020-12-23:
Send me a test email to dicky.tech@gmail.com I will forward you the documentation from Datecs fp300